sebanc/brunch #1586: Error in installing linux ( chrome OS Version 102.0.5005.125 )

nameismohitkumar, 2022-07-13:
Linux is not installing in My laptop, I have tried a lot but it is always showing the same message " Error installing Linux ". I saw old solutions for this in which they are saying to disable crostini....(.something something ) but it is not showing in my chrome://flags. �
[image]
�

SuRmsu, 2022-07-14:
> Try enabling "enable updates" in the Brunch framework options. Just enable this option. Do not actually upgrade to 103 but stay on 102. See https://github.com/sebanc/brunch/blob/master/troubleshooting-and-faqs.md for information how to set the enable updates option.

It works. Don't upgrade Brunch by mistake

Jeweldation, 2022-07-14:
> It works. Don't upgrade Brunch by mistake
What happens if you upgrade?

block6791, 2022-07-15:
This will happen: https://github.com/sebanc/brunch/issues/1574

nameismohitkumar, 2022-07-17:
@block6791 thanks :)

closed 2022-07-17 by nameismohitkumar