sebanc/brunch #1593: login loop + black screen after login

lavilao, 2022-07-23:
Hi I currently have installed chromeos rammus 103 stable with brunch r103 unstable 20220715 on a Asus x453sa laptop (n3050 Celeron) and 1/2 of the times I boot into chromeos it Will ask me for password go to a black screen go back to login write password agan it logs in and after a while the screen Will go black with only the cursor showing (not moving). The only way of get out is with a hard shutdown, the system is running in this state because I can go to tty but Even there I can't do anything because I. Not un developer mode ando when I write anything to the user and press Enter the screen Will go black with a white blinking square at the top left part. Anyone can help me?

lavilao, 2022-07-26:
Fixed by using 5.15 keenel

closed 2022-07-26 by lavilao

Bwangster12, 2022-12-04:
I seem to be having this issue and am using the 5.15 kernel.  It seems that random changes will cause it.  I've added the ACPI_Power_Button and that caused it once and then this morning tried adding a 2nd account and that caused it.