sebanc/brunch #1594: Suggestion: Kernel selection menu in script

dazbeattie, 2022-07-24:
Firstly, thanks for a great project!

I am running Brunch on a number of devices, mostly Surface Pros. I can only get Brunch to work if I force the kernel to be at 5.4. Wondering if a kernel selection menu could be added, so that the kernel is correctly set at the time of build ?
If no go with this, all good, just a suggestion!

Keep up the great work !!

sebanc, 2022-07-24:
There was a bug a few releases back which prevented surface devices to work with kernel 5.10 and 5.15 but it has supposedly been fixed.

Have you tried those kernels in recent brunch versions ?

dazbeattie, 2022-07-31:
Hi @sebanc. Thanks for the response. I have tried to use 5.15 on several different Surfaces (6) and end up with a boot loop everytime. Switched back to 5.4 and everything works

dazbeattie, 2022-12-02:
Hi @sebanc. Just took a fresh install on another Surface with the latest Brunch stable. Left it on default kernel 5.15 and all booted up fine. Went back  to my original machine and changed to 5.15, all working fine.

closed 2022-12-02 by dazbeattie