sebanc/brunch #1600: Something Problem on my ASUS BR1000FKA

Wd2wad3, 2022-07-31:
I install brunch r103 and dedede recovery on my notebook(+CPU is Celeron N4500).
wifi, touchscreen, speaker, camera, and something other is working well.

But, Internal Microphone isn't work, and trackpad isn't work well after wake up(+this problem is on Ubuntu 22.04 too).

Can you give me some solution, @sebanc ?

tejasraman, 2022-08-09:
Hi @Wd2wad3 !
Try some framework options. (Internal mic fix and try all 3 trackpad-resume fixes) - trackpad resume fix #2 fixed it for a laptop that I installed brunch on but it really depends. 

To access the config menu, open crosh (Ctrl+Alt+T), type shell, and than type "sudo edit-brunch-config" to access config menus.

4rNe5, 2023-09-06:
WTF....I didn't Close This issue. Thank You For Your Help!!

closed 2023-09-06 by 4rNe5

tejasraman, 2023-09-06:
😀

4rNe5, 2023-09-07:
> 😀
thanks Bro!