sebanc/brunch #1601: External monitor flickering

bdodroid, 2022-08-03:
Just did a fresh install of release 103 and volteer 103 and I'm having issues with my external monitor.

It will go black for a half a second then be fine for a couple seconds then repeat.
Tried 3 different adapters/cables and the issue persists. I even booted back into my windows 11 partition (I setup dual boot) and verified the external monitor works just fine with it.

The weird thing is it almost seems triggered by my mouse movement. While just typing this message I had zero flickering but the second I move my mouse the screen started flickering again.

PC: Asus Flow Z13 ( 12th gen i9 version) - onboard intel graphics but it also has gtx3050 mobile gpu that isn't active as far as I can tell.
Brunch: r103 stable 20220721 (dual boot with grub2win)
ChromeOS: Volteer 103.0.5060.132 (Official Build)

Anshu-gits, 2022-08-07:
I had the same problem with my HP. It was fixed when I reset the chromebook

bdodroid, 2022-09-12:
Seems like an issue with brunch and 12th gen cpu's.

I have another 12th gen i7 laptop and it has the same issue when connected to an external display but both my 11th gen devices are totally fine and if I install the latest ubuntu or fedora versions my external monitor works fine as well. 

I initially thought it was an issue with having a discrete gpu but my second 12th gen device doesn't have one.

bdodroid, 2022-12-12:
So FydeOS is working with an external monitor on my 12th gen laptop but I cant get a version of brunch to support it. 

Not sure what they have done to the build to make it work and they are even using 5.10 kernel (5.10.133-15812-gbb55053cbec3-dirty)