sebanc/brunch #1604: Blask screen after first setup

Jstnldrs05, 2022-08-05:
Good day everyone, can you guys help me with my issue.

I recently installed Chrome OS in my laptop with brunch 103 stable and a 103 image by Zork and 15.15 ver of kernel. I am experiencing black screen issue after I setup my google account. After the setup, the Desktop/GUI of the OS opens for one second and then it blacks out my screen. Which I thought that it's just loading the system but it's not and I need to force restart my laptop to open it again and it boot loops. So, I build another new OS to set it up again.

After several trial and errors, when my screen is black, I try to open the developer console (ctrl alt F2) and it turns on my screen and opens the terminal which means that it is not the backlit of the screen. Maybe the main GUI of chrome OS was the problem.

I also, try to change kernel version to 15.10 and it successfully installed with no black screen issue, but the problem is my trackpad doesn't work while in brunch 15.15 version my trackpad works flawlessly. I also try to config brunch for my trackpad and none of them solves the problem.

My main concern here is the Desktop/GUI which is black on my screen on kernel 15.15 version
While on the kernel 15.10, the Desktop/GUI displays smoothly, but the trackpad won't work

I am using Lenovo Legion 5 15ARH05H
CPU: Ryzen 7 4800h with integrated graphics
GPU: GTX 1660ti
RAM: 16gb

Hopefully you can help me about my problem. Thanks to all.

tejasraman, 2022-08-09:
Try kernel parameter `nomodeset`

Jstnldrs05, 2022-08-23:
> Try kernel parameter `nomodeset`

how to set it up? I' so sorry, i don't have any idea about it. I appreciate your response.

tejasraman, 2022-08-23:
I can give you some instructions:
1. Click the down arrow and enter on the boot screen to select "ChromeOS (Settings)"
2. Click Enter and get to the screen with custom kernel parameters
3. Type `nomodeset` and press enter

And you probably know the rest!
Actually, I had this issue. Nomodeset might resolve it, but personally 5.10 fixed a black screen after login issue on my Ryzen mini PC.

tejasraman, 2022-08-23:
@Jstnldrs05 Don't try nomodeset - UPDATE

tejasraman, 2022-08-23:
My new advice is to use [this](https://gist.github.com/jbuncle/7dacde983b3c33b3b816b10e2fd2308a) patch on 5.10. (Nomodeset breaks chromeOS)

tejasraman, 2022-09-16:
That patch won't work on brunch. The 5.15 no desktop is more concerning. @sebanc help?

closed 2026-01-13 by github-actions[bot]