sebanc/brunch #1608: Storage is using 60 gb in system

Anshu-gits, 2022-08-07:
@sebanc

tejasraman, 2022-08-09:
Huh?
We need more context. What storage? How much space does your HDD have? And are you using single or dualboot?

Anshu-gits, 2022-08-09:
I have a nvme of 256gb and am using it in single boot mode. The storage taken my system is now 77gb
[image]

tejasraman, 2022-08-09:
Ouch...
It should only take a few GB...
Reinstall maybe?

What did you do to it?

DTayA1, 2022-08-14:
It may just be the fact that this is running within an img, CrOS doesn't seem to mesh well with certain sizes, but 3rd party file managers etc usually detect the proper size even when the OS doesn't. Try CX File Manager, it detected the proper storage size for me.

tejasraman, 2022-08-14:
If you installed in singleboot, an img is not used.

tejasraman, 2022-08-14:
@DTayA1 are you using dualboot?

Anshu-gits, 2022-08-15:
No, it was a single boot

closed 2022-08-15 by Anshu-gits

tejasraman, 2022-08-15:
@Anshu-gits I know that it was a single boot because you said that earlier.

> I have a nvme of 256gb and am using it in **single boot mode**. The storage taken my system is now 77gb [image]

@DTayA1 was obviously using it in **dualboot**.
The best advice I can give you is to reinstall Brunch using the guide in this repo.