sebanc/brunch #1617: What's up with the CPU support?

stxrrynight008, 2022-08-15:
ChromeOS Flex seems to support CPUs older than Sandy Bridge without issues (e.g. my ThinkPad L512 with i5-540M) but brunch doesn't.
Is this a special thing for ChromeOS Flex or is it an issue with brunch? Please let me know. Thanks!

sebanc, 2022-08-15:
In order to optimize chromebooks performance, chromeos recovery images are built with specific cpu instructions which target the end device cpu whereas chromeos flex is built with a limited set of cpu instructions to support more devices.

stxrrynight008, 2022-08-16:
Alright, thanks! Before I close this, do you know any alternatives to ChromeOS with Play Store that will work on legacy devices? I have tried FydeOS but that turns out to only support GPT.

closed 2022-11-13 by stxrrynight008

Anurag1061, 2024-09-22:
Is Intel Kaby Lake Supported Core-i5 7th gen (Kaby Lake) Running r120 well ?