sebanc/brunch #1618: Problems when installing Chrome OS to my laptop

thisisGS2010, 2022-08-16:
Hello.

I am trying to install Chrome OS in my `LG LAPTOP 15U34`.
The USB memory stick I used is `Sandisk Cruzer Blade 16GB`.

Here's my progress:
    * Used `Rammus Leona` version `101`.
    * The version of Brunch Framework is `Brunch r103 stable 20220721`.

  I installed the `chromeos.img` file on my USB memory stick, and I set the BIOS to boot Chrome OS from USB.

And then, I started Chrome OS setup.
I selected the language, connected to the WIFI.
And then here's the problem.
The screend asked "Add who to the chromebook?"
I selected `Me` and proceeded.
And then, a pinwheel saying `please wait` spins for while, and the screen prints out an error `Oops, something went wrong with signing in Please connect to the internet to sign in to your chromebook`
I definently connected to the WIFI!

So after this, I pressed `CTRL`+`ALT`+`F2` to enter the terminal, and logged into `root`.

I typed:
    `sudo edit-grub-config`
But I can't get into the NANO text editor, instead I entered to Brunch configuration menu.

By the way, is selecting the kernel option `iwlwifi_backport` effects on WIFI?

I tried `powerwash` and `powerwash and revert` couple of times. In Chrome OS

Also, my CPU is `intel pentium`:I don't know what generation it is.

Is there anyone can solve my problem?

thisisGS2010, 2022-08-16:
_It will be thankful when you answer ASAP_

rayep, 2022-08-28:
Same as https://github.com/sebanc/brunch/issues/1549

closed 2026-01-13 by github-actions[bot]