sebanc/brunch #1625: stuck in sign-in

quark365, 2022-08-18:
I installed chromoes version 103 in legacy mode laptop using mbr patch but it is blank when we have to type the email id what to do . ca

https://user-images.githubusercontent.com/102899521/185378707-d8ebc83a-3b31-4f97-91e5-b772baab08b1.mp4

quark365, 2022-08-18:
HP Pavilion dv6, intel i5 (5th gen ig) , 6gb ram

quark365, 2022-08-18:
similar issue in #1618

quark365, 2022-08-18:
chromeos version 99 is working fine but crosh won't open , so I installed version 90 and updated brunch using crosh , but on editing grub-config the to use enable_updates , the code was completely different from others.

quark365, 2022-08-18:
even after updating to latest version 104, the same issue occurs  what to do ?
is it because of chromeos verification or something. if so to disable it refresh and esc button is to be pressed ryt while booting, but which key is refresh key in normal keyboard 

please find a fix for this issue

quark365, 2022-08-18:
and crosh is not loading in legacy mode for latest versions

Suleman-Elahi, 2023-04-14:
Happening the same thing today on my i3 3rd gen laptop. Image 103.

Arunkapil01, 2025-04-24:
@Suleman-Elahi hello which brunch you used on intel i3 3rd gen

ilovefreesw, 2025-04-30:
> [@Suleman-Elahi](https://github.com/Suleman-Elahi) hello which brunch you used on intel i3 3rd gen

@Arunkapil01 Sorry I do not remember .... but I moved to FydeOS.