sebanc/brunch #1628: Touchscreen not working on Surface Laptop 3

rikkert, 2022-08-18:
I sucesfully installed r103 on my Intel SL3 and it is working great.
Now I'm trying to get the touchscreen to work but it is not responding at all.
Here is what I have done so far: enable ipts on kernel 5.15. This seems to have no effect, also tried r104 with the same result.

dmesg output related to ipts:
`
[    1.372393]     options=enable_updates,pwa,ipts_touchscreen
[   30.289835] brunch: /rootc/patches/99-ipts.sh ipts enabled
[   30.494282] brunch: /rootc/patches/99-ipts.sh success
[   33.676433] ipts: loading out-of-tree module taints kernel.
[   33.678475] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[   33.680535] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Device 045E:0983 ready
[   33.746143] init: ipts main process (2500) terminated with status 127
`
when I run ipts-dump I get this:
`
$ sudo ipts-dump
ipts-dump: error while loading shared libraries: libfmt.so: cannot open shared object file: No such file or directory
`

sebanc, 2022-08-18:
Could you try with the below brunch-unstable build ?
https://github.com/sebanc/brunch-unstable/releases/tag/r104-unstable-20220818

rikkert, 2022-08-19:
Thanks, I just upgraded but after a reboot the screen still doesnt respond to touch.
The dmesg output did change a bit:

`
[   14.297959] ipts: loading out-of-tree module taints kernel.
[   14.301588] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[   14.304518] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Device 045E:0983 ready
[   14.370029] init: ipts main process (2162) terminated with status 1
`

I can run sudo iptsd without the dependency error.
Hopefully this helps and can be fixed. I also tried to use kernel4 but that doesnt boot for me at all.

sebanc, 2022-08-19:
Do you have any error message when you run "sudo /usr/bin/iptsd" ?

If the ipts daemon does not exit immediatly, is touch working ?

rikkert, 2022-08-19:
`
$ sudo /usr/bin/iptsd
[13:16:44.943] [error] You need to specify the hidraw device!
`

Not sure what to specify, can't really find anything about that.

rikkert, 2022-08-19:
When I try to identify the hidraw devices the device detected by the kernel driver is not there (Device 045E:0983 ready):
```
hidraw0          Microsoft Surface 045E:0922
hidraw1          Microsoft Surface 045E:09AE
hidraw2          Microsoft Surface 045E:09AF
```

sebanc, 2022-08-22:
Actually Surface laptop 3 should require "ithc_touchscreen" option, could you try it ?

rikkert, 2022-08-22:
Thank you so much, that was it! 
All works now perfectly. 
Didn’t imagine the laptop was new enough, not sure where I should’ve found out.

closed 2022-08-22 by rikkert