sebanc/brunch #1629: Sound and brightness problem on Lenovo 11e Chromebook 4th Gen

breghar, 2022-08-19:
Hello,

I managed to successfully install Brunch on the previously mentioned notebook (I only had to replace boot.efi which was mentioned somewhere here in the issues). Unfortunately sound is not working (board is pyro and intel generation apollo lake). I tried different settings - while going through the doc I figured best would be kernel 5.4 chromebook, enable HT, pw, native image, disable dsp etc. I tried also with other configurations (without native image, standard linux kernel, older kernel) - no luck. I got sound working on 5.15 kernel on GalliumOS (by updating kernel, installing new sof bin firmware and then doing patch - options snd-intel-dspcfg dsp_driver=3
options snd-sof-pci fw_path="intel/sof" ). I assumed as its native hardware for chromeos and im using native recovery image (pyro) it should be working without hacks? I think that before changing settings display brightness worked, but now its also not working. And battery life is much worse than on Gallium/stock Chrome OS, so I am not sure about power management (?). 

Any chance to point me in the right direction in how to fix this?

breghar, 2022-08-20:
Just as an update - managed to get sound working, but without mic. If anyone has similar problem -> I stayed with default 5.15 kernel and without native-chromebook flag. Those two in my case seemed to work worse (e.g. setting brightness was broken, reboot while going to developer console etc.). 

I enabled sound with https://www.reddit.com/r/chrultrabook/comments/uc0b6i/howto_audio_on_apollolake_devices_under_linux/ . Mainly modprobe + copying tplg. Volume scaling works fine. Mic is also showing and I can change volume but it does not record anything on any app. I went through whole 'mic not working' threads but amixer internal_mic did not work. On GalliumOS I managed to get it working by adding load-module module-alsa-source device=hw:0,99 to pulse audio config. Here I am struggling a bit to do that. So again - if anyone can jump start me a bit I would highly appreciate it.

sebanc, 2022-09-01:
You could try pyro recovery + kernel 4.4 + native_chromebook_image option.

breghar, 2022-09-03:
This is what I started with but it did not work. Was even worse (screen brightness control stopped working)