sebanc/brunch #1632: Touchscreen kernel commands no longer working in r103?

clee604, 2022-08-26:
Hi there,

I'm not sure if there is a regression in the new releases or if there's something that now breaks the kernel command to invert the touchscreen. Previously using r100/chromeos100 I would use 
of_touchscreen.custom=1 of_touchscreen.invert_x=1 in the brunch kernel config to fix my touchscreen. 

In the new release, this doesn't seem to work anymore, it's the same with or without the extra parameters. Does this kernel configuration no longer work?

sebanc, 2022-09-01:
Ah indeed, could you try with "input-core.custom=1 input-core.invert_x=1" ?

or "touchscreen.custom=1 touchscreen.invert_x=1" if the first suggestion does not work ?

clee604, 2022-09-04:
"input-core.custom=1 input-core.invert_x=1" worked! Thanks!

closed 2022-09-04 by clee604

clee604, 2022-09-04:
maybe more of a feature request, but is there the possibility to put touchscreen x and y inversion as configuration options? or at least a way to bring up/select previously used custom parameters when entering the configuration?