sebanc/brunch #1633: R103 system storage bumped up to nearly 50G

javacom, 2022-08-26:
For R102 the storage size of System was only 20G, but R103 would bumped up to nearly 50G and increasing, is there anything wrong here?

v-mod, 2022-09-19:
I seem to be having the same issue

raduburla, 2022-09-25:
Maybe there are a lot of leftovers from the update? I did update Brunch and ChromeOs at the same time. After brunch update I did a powerwash because it was required but didn't do after ChromeOs (upgrade from 97 to 105)

A new powerwash after ChromeOs upgrade fixed the issue for me. 

Before system had ~45GB
After powerwash ~20GB

v-mod, 2022-09-25:
I have already power washed when switching from beta to stable channel so I
don't think that's going to help.