sebanc/brunch #1642: No internal sound Steam Deck Running Chrome OS

MrHomebrew, 2022-09-04:
The internal sound seems to be not working on the steam deck running ChromeOS. HDMI/external audio does work fortunately.

Is there any chance there could be a fix for the internal audio on the Steam Deck itself?

RibhuNoodleBoodle, 2022-10-21:
What about stuff like Hardware Acceleration and android support?

tejasraman, 2022-10-21:
What kernel and recovery image are you using?

tejasraman, 2022-10-21:
You need a newer kernel than 5.15.

MrHomebrew, 2022-12-24:
Is there a newer kernel that can be used?

tejasraman, 2022-12-24:
Not easily....

MrHomebrew, 2022-12-26:
I would definitely be willing to test, if we could get it done it would really up the compatability for the device. Everything else seems to be working good, just the internal audio issue.

Unless there is another way around to add audio compatability for the steam deck, some kind of hackers maybe.

Seeing as it is a device that has the same hardware for all users the Steam Deck would be the perfect device to support.

guylamar2006, 2022-12-31:
What kernel/boot options and brunch recovery are needed to boot on the steam deck? Wanna install this on my Steam Deck.

tejasraman, 2022-12-31:
Try the zork image….

guylamar2006, 2022-12-31:
What's the setup on the brunch setup screen? Boot options/kernel version/kernel options?

tejasraman, 2022-12-31:
Idk but try 5.15 and vanilla to start

guylamar2006, 2022-12-31:
I finally got brunch to boot after having bootloops with brunch-unstable. Comfirmed zork, I was pretty sure it was zork since the Steam Deck's cpu is built on AMD's Zen 2 technology. (AMD Van Gogh/Custom APU 0405)

Brunch Setup Options:
Recovery: Zork 107 from [Here, Search for zork](https://chromiumdash.appspot.com/serving-builds?deviceCategory=ChromeOS)
Kernel: 5.15
Brunch 107 Stable
Options:enable_updates,pwa,sysfs_tablet_mode,force_tablet_mode
Kernel Options:enable_hyperthreading=1

My settings after trial and error:

Settings > Language And Input > Show input options on shelf
Accessibility > Keyboard > Show on-screen keyboard

Display >  Display Size > 1280x800 for decent appearance. (All the way to the left).

Confirmed that everything else works but sound from the speakers, Bluetooth audio is only workaround for now until native audio support gets into the kernel.

tejasraman, 2022-12-31:
I know, but I am trying to backport Kernel 5.19 to brunch in hopes that it will still work.

tejasraman, 2022-12-31:
I know, but I am trying to backport kernel 5.19 to brunch in hopes that it will still work properly... There are still a few issues I want to sort out, and 5.19 isn't an LTS so I'm sure Brunch will never officially include it, but I'm hoping that 6.1 (which is an LTS) can vent be ported.

MrHomebrew, 2023-01-10:
whenever you have a chance to send a test build to test be sure to let us know. i will also give it a test run.

I am using kernel 5.15 also, brunch stable 107.

Unstable brunch 108 build leads to a boot loop.

Resolving the internal audio issue be amazing but not easy obviously.

tejasraman, 2023-01-11:
UPDATE: I'm building a build with the UCMs for ACP5X on my GH Actions page. I will send the test build as soon as it's done.

Keep in mind that this is strictly a _test_ build. Once this is merged (if it works) than I will tell you to not use the test build anymore.

Also please keep in mind that if the sound still doesn't work try running `alsaucm -c acp5x set _verb HiFi set _enadev Speakers` or `alsaucm -c acp5x set _verb HiFi-upstream set _enadev Speakers`

MrHomebrew, 2023-01-13:
Ready to test when you are good to go.

Just wondering are you testing this with brunch stable or brunch unstable?

guylamar2006, 2023-01-13:
Stable, unless the bootloop is fixed with the deck.

guylamar2006, 2023-01-13:
If the bootloop is fixed I would love to test the unstable branch and report any issues since it gets updated quite a bit.

tejasraman, 2023-01-13:
Can't build. Submitting it to the unstable repo instead.....

tejasraman, 2023-01-14:
See the status of the [brunch-unstable PR #9](https://github.com/sebanc/brunch-unstable/pull/9) where I submitted the ACP5x UCMs. This is to be used with 5.15

MrHomebrew, 2023-01-17:
Last time I tried unstable brunch there was still a boot loop on the Steam Deck. Not sure if anything has changed recently, will have to test and see

MrHomebrew, 2023-01-22:
@tejasraman I was able to build with the Steam Deck audio changes. No sound yet

I tried

chronos@localhost / $ alsaucm -c acp5x set _verb HiFi set _enadev Speakers
ALSA lib /build/zork/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/ucm_cond.c:207:(if_eval_control_exists) cannot determine control device
ALSA lib /build/zork/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:963:(snd_use_case_mgr_open) error: failed to import acp5x use case configuration -22
alsaucm: error failed to open sound card acp5x: Invalid argument

chronos@localhost / $ alsaucm -c acp5x set _verb HiFi-upstream set _enadev Speakers
ALSA lib /build/zork/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/ucm_cond.c:207:(if_eval_control_exists) cannot determine control device
ALSA lib /build/zork/tmp/portage/media-libs/alsa-lib-1.2.1.2-r2/work/alsa-lib-1.2.1.2/src/ucm/main.c:963:(snd_use_case_mgr_open) error: failed to import acp5x use case configuration -22
alsaucm: error failed to open sound card acp5x: Invalid argument

tejasraman, 2023-01-22:
Are the files inside the directory? Take a screenshot with `ls` and show me please

MrHomebrew, 2023-01-22:
Yes they are but let me get a picture

MrHomebrew, 2023-01-22:
Here is a picture

[image]

MrHomebrew, 2023-01-22:
Did not mean to close this as it is not fully resolved.

tejasraman, 2023-01-22:
It's ok... We just need a kernel patch to support it as well as those UCMs

MrHomebrew, 2023-01-22:
I'm able to build it now so if we can get that kernel patch I can proceed to the next step for sure.

Appreciate your time and help, you'll definitely get credit for this in the community.

tejasraman, 2023-01-22:
How?

I've been trying but GitHub actions building is failing every time and I don't want to leave my PC on all night.

There's a file in the unstable branch (in the kernel patches 5.15 directory) where you can add build parameters.

I'll try to build brunch with only 5.15 to test and report back. If you can build it that's great. Could you build it??

MrHomebrew, 2023-01-22:
Yes I built with all the kernels.

Also made sure to built with your changes on the stable branch to prevent bootloop issue with brunch unstable

MrHomebrew, 2023-01-22:
@tejasraman if you can make the changes for this kernel patch on your unstable branch fork, I can put them in my stable fork. Build and see if it works with the kernel patch. Save you some time as my computer can finish compiling in less then 2 hours.

tejasraman, 2023-01-22:
I am adding them to my stable fork.

Tejas

MrHomebrew, 2023-01-22:
Sounds great, I will keep an eye out for the changes

tejasraman, 2023-01-22:
I added them to [https://github.com/tejasraman/brunch](https://github.com/tejasraman/brunch)

MrHomebrew, 2023-01-22:
Will start the build in a few minutes and test right afterwards

MrHomebrew, 2023-01-22:
Think I may update the build instructions on my branch also at some point. Help you be able to compile locally in the future

tejasraman, 2023-01-22:
Tysm @MrHomebrew 👍

MrHomebrew, 2023-01-23:
@tejasraman still no internal sound and tried the commands from before.

Anything e;see you can think of?

tejasraman, 2023-01-23:
Can you run ls again and show me the output of the commands in a photo?

MrHomebrew, 2023-01-23:
@tejasraman 
Here you are

[image]

tejasraman, 2023-01-23:
Did you update or reinstall the brunch on the steam deck yet? What's the kernel version? Did you use my repo to build?

MrHomebrew, 2023-01-23:
I made a new chromeos.img with the new brunch that had the changes you recently made on brunch stable. Which included the kernel patch.

uname -r
5.15.73-brunch-sebanc

tejasraman, 2023-01-23:
@MrHomebrew could you tell me the output of this command (`lspci -v | grep Audio`)?

Also, I want to try another kernel config that is the one I was supposed to enable. Could you build again?

MrHomebrew, 2023-01-23:
Will start the build and get you that info asap

tejasraman, 2023-01-23:
Sorry if this is taking too long @MrHomebrew

MrHomebrew, 2023-01-23:
Not at all happy to help any way I can.

Started the new brunch build

MrHomebrew, 2023-01-23:
localhost / # lspci -v | grep Audio
04:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 1640
04:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor (rev 50)
localhost / #

MrHomebrew, 2023-01-23:
Still no sound at the moment.

Finished latest brunch build.

May have to continue in the morning.

Will get it eventually.

Yougli, 2023-05-13:
Hi,

I'm trying to boot my Steam Deck with ChromeOS via Brunch, but the boot manager doesn't find my USB or my microSD card...

I've followed the official instructions (here: https://github.com/sebanc/brunch/blob/main/install-with-linux.md) with and without the "Special instructions for Legacy devices only", both on my USB drive and my microSD card.

I'm using Zork r111 recovery image, and Brunch r111 stable.

Since you guys were struggling to get the internal audio to work, I figured you at least succeeded in booting to ChromeOS, is there something I'm missing?

Thanks in advance for your input!

MrHomebrew, 2025-02-09:
Well I must say with the latest Brunch r132 unstable 20250209 that I tested audio internal audio speakers seem to be working on the Steam Deck LCD 😉 great work!!!!

[image]

closed 2026-01-12 by github-actions[bot]