sebanc/brunch #1654: v4l2loopback and ms surface go 2

Lucaacer, 2022-09-13:
Good afternoon Sebanc, I recently installed chromeos to my ms surface go 2, pyro recovery and it works very well but for the 2 cameras, as expected.

I supposed that brioche would be a viable workaround to get cameras working as v4l2loopback devices through libcamera (they do work in ubuntu for me). I can't built libcamera, though, because I can't find a way to download the headers for you kernel. I mean that v4l2loopback.ko is missing and I am not able to build it  neither from source, nor by simply installing the libcamera-dev package, due to the missing headers of your custom kernel.

Please, have you got any suggestion?

thomasjungblut, 2022-10-27:
Installing libcamera was a bit of a struggle (on my surface go 1 with rammus), but I got there.

First get chromebrew to install packages: https://github.com/chromebrew/chromebrew 
Then you can install pretty much all the stated packages (meson, ninja, cmake, binutils etc) from https://libcamera.org/getting-started.html 

The linux headers are just a simple `crew install linuxheaders` away.

Lucaacer, 2022-10-30:
> Installing libcamera was a bit of a struggle (on my surface go 1 with rammus), but I got there.
> 
> First get chromebrew to install packages: https://github.com/chromebrew/chromebrew Then you can install pretty much all the stated packages (meson, ninja, cmake, binutils etc) from https://libcamera.org/getting-started.html
> 
> The linux headers are just a simple `crew install linuxheaders` away. This successfully compiles libcamera itself albeit without gstreamer, which I haven't found a solution so far. So the loopback might still be unresolved, but I haven't tried yet.

Sorry, I seem to get you are using crostini, while I am using brioche https://github.com/sebanc/brioche

Since crostini doesn't have any direct access to hardware and uses its own kernel, which is different from the one of chromeos, and it doesn't allow you to add new modules, I don't think the v4l2loopback.ko might be inserted in any way. It might be compiled, but there'd be no way to use it.

Regarding me, since brioche in nothing more than a chrooted linux (with the nifty addition of chromeos composer instead of xorg as for crouton) and it uses brunch kernel which has got access to the bare metal, I was trying to build v4l2loopback.ko for that kernel, not for the one of crostini. All the required software is avalaible, gstreamer included. The only issue is that the module must be compiled against the exact kernel and I did not know where to find the headers of Sebanc's one. Maybe I could use the generic headers for that revision of the kernel, but I haven't tried so far.