sebanc/brunch #1655: Using partitioning tools in Windows will brick Chrome OS

Maxwellads, 2022-09-14:
I made my device a dualboot of Windows + ChromeOS, but not by the method suggested by brunch guide (creating an chromeos image and add boot entry by grub2win). Instead, I installed it wiping entire disk first, and resized H-STATE, installed windows, added an chain entry to jump from grub2 to windows boot loader.

I figured out something weird - whatever I use for partitioning, if not in linux environment, will brick the grub boot entry. It appeared that the partition number has changed, for example, ROOT-C WAS (hd0,7), but after that I added an NTFS partition on the end of my drive, it became (hd0,3).

I edited grub entry to the corresponding part. Unforunately, it worked but not really. After loading brunch, it will be thrown into a reboot.

Can somebody tell me how to fix this, and how I can install without wiping the whole disk? I already wiped for 10 times or so...

ameyjoshi3068, 2022-10-08:
Using partitioning tools in windows many times mess up with chromeos. Here is what I did

1. Wiped disk and installed windows
2. Installed lite linux os (Just wased 20gb space)
3. Installed chromeos from linux in seperate ext4 partition

Just make sure to disable quick boot in windows and use ONLY gparted for partition management

Maxwellads, 2022-10-08:
> Using partitioning tools in windows many times mess up with chromeos. Here is what I did
>
> 1. Wiped disk and installed windows
> 2. Installed lite linux os (Just wased 20gb space)
> 3. Installed chromeos [...]

Same as the procedure I am following now. Just that formatting any partition on the same disk will brick it too (which sucks)

tejasraman, 2022-10-21:
Use the "normal" dualboot. IME windows never messes up if you use ".img style" dual boot

Maxwellads, 2022-10-22:
I hoped so, until it eventually messed it up when I created an NTFS partition in Windows. Thanks for suggestion though.

tejasraman, 2022-10-22:
Interesting.....

closed 2026-01-12 by github-actions[bot]