sebanc/brunch #1656: Volteer - Images are down

v-mod, 2022-09-16:
When I check for the Chrome OS Recovery images on cros.tech it says:
`Google's data source is not currently providing all recovery images.
You may be able to find them [on Google's site](https://chromiumdash.appspot.com/serving-builds?deviceCategory=Chrome%20OS) until they fix the issue.`

TheMCNerd2017, 2022-09-19:
Google's site doesn't seem to show recoveries either (or I'm looking in the wrong place). Also this is happening for every device on cros.tech, not just Volteer.

v-mod, 2022-09-19:
A way I have found to get the latest images is to download the latest recovery.json file from Google: `https://dl.google.com/dl/edgedl/chromeos/recovery/recovery.json` and then to search that for the model i need, in this case volteer.

albsrocha, 2022-09-26:
> A way I have found to get the latest images is to download the latest recovery.json file from Google: `https://dl.google.com/dl/edgedl/chromeos/recovery/recovery.json` and then to search that for the model i need, in this case volteer.

here -> https://chromiumdash.appspot.com/serving-builds?deviceCategory=ChromeOS

v-mod, 2022-09-26:
What exactly do you mean:
[image]

TheMCNerd2017, 2022-09-27:
> here -> https://chromiumdash.appspot.com/serving-builds?deviceCategory=ChromeOS

That site only shows download links if you're running a desktop OS (I just found this out now when accessing it on my PC which I've not done before). On a mobile OS like Android only the first four columns are shown, even with desktop mode enabled and using landscape orientation. Additionally the 'Download CSV' button is hidden as well.

albsrocha, 2022-09-28:
on this site you download the current and old images. the print you posted, the link is clear.

albsrocha, 2022-09-28:
I accessed it on mobile and it worked

TheMCNerd2017, 2022-09-29:
> I accessed it on mobile and it worked

Good that it works for you. I accessed it again and it still doesn't show up correctly, even with desktop mode enabled. I'm guessing something with the browser or mobile OS I use is causing the table to be truncated and CSV download option to be missing. Below is what the site appears as on my phone.

<details><summary>Site</summary>

[image]

[image]

</details>

Worse case I'll just grab the direct links from my PC and send them to my mobile device.

v-mod, 2022-10-09:
> > I accessed it on mobile and it worked
> 
> Good that it works for you. I accessed it again and it still doesn't show up correctly, even with desktop mode enabled. I'm guessing something with the browser or mobile OS I use is causing the table to be truncated and CSV download option to be missing. Below is what the site appears as on my phone.
> 
> Site
> [image]
> 
> [image]
> 
> Worse case I'll just grab the direct links from my PC and send them to my mobile device.

Have you tried using your phone in landscape?

TheMCNerd2017, 2022-10-21:
> Have you tried using your phone in landscape?

[image]
Sorry for the very late reply, but the same thing happens in landscape mode as well.

v-mod, 2022-10-21:
> [image] Sorry for the very late reply, but the same thing happens in landscape mode as well.

Yes, thanks for that. 
Interesting. If you go on chrome dev tools and change the dimensions it causes that. I think it probably does have something to do with the fact that the screen size is causing the table to be truncated.