sebanc/brunch #1666: Brunch R103 ==> Loading Bruch Framework loop

dneusitz, 2022-10-03:
Just upgraded to RAMMUS 105.0.5195.134 using Brunch r103 stable 20220721
results in the boot loop ==> Loading Bruch Framework...

Saw a similar issue and tried the unstable R103 220720, but same result.
Let me know what doc you need to diagnose.
Thanks

ameyjoshi3068, 2022-10-08:
Try latest Brunch r105 unstable 20221005 + chromeps 105

dneusitz, 2022-10-11:
@ameyjoshi3068 

Thanks for responding.
Where can I find Brunch r105 unstable 20221005?

ameyjoshi3068, 2022-10-11:
Here https://github.com/sebanc/brunch-unstable

dneusitz, 2022-10-13:
Sorry
I do not understand. is there a specific file name or folder that I should be looking for in the link you gave me? @ameyjoshi3068

ameyjoshi3068, 2022-10-14:
Just download brunch unstable release from *release* section

Since you have no acess to terminal in chromros the easiest way is to reinstall ChromeOS using 105 rammus and brunch unstable 105

closed 2022-10-17 by dneusitz