sebanc/brunch #1671: Google account not logged in

j-liaman, 2022-10-08:
lenovo g500 laptop
specification
CPU: Pentium 2020m
GPU: IGPU
RAM: 8GB DDR3
SSD 128GB
After selecting a personal account, I want to log in.
Please wait no further progress
What should I do

ameyjoshi3068, 2022-10-12:
Have you download Chrome OS 103 with latest brunch?

tejasraman, 2022-10-15:
@ameyjoshi3068 can you explain what this generic response is? Downgrade ≠ fix.

tejasraman, 2022-10-15:
Hi @j-liaman could you please try octopus 103 just for my reference? This is obviously not a _you_ problem (see #1674) and I've asked the same to the people there.

Tejas

ameyjoshi3068, 2022-10-15:
> 

To be honest I also think the same.

ameyjoshi3068, 2022-10-15:
Now coming to problem,The exact same problem is also happening with me. in my case cpu is Pentium 4th gen. I tried many things on my own but can't solve it. the only way I got my mechine working is by downgrade so I thought it should be worth to mention this.

tejasraman, 2022-10-18:
Hi @ameyjoshi3068 could you try Caroline or Octopus 105? I know that Caroline works with sandy bridge (with 105) and Haswell is newer so it should work... 

The problem is that Caroline only gets updates until 2024.....

j-liaman, 2022-10-20:
I used Rammus up to version 100
I'm trying to install a new version 101, but there was a problem with the Google sign-in process. It doesn't work from version 101 or later.
Also, there was a problem while using version 100, but crosh was opened.
immediately shuts down

ameyjoshi3068, 2022-10-20:
I tried caroline and octopus but none of them solved sign in issue.

Nyta8591, 2023-06-03:
I also have Google login issue with this laptop https://chromebookdb.com/chromebook/hp-chromebook-14-14-ak040nr/ . Its EOL as of 2021 and latest OS was 93 i think. Anyone has 93version for Brunch and recovery before 100 i could try? 🙏

also audio and bluetooth doesnt work, and framework options werent able to fix audio. I managed to pull audio drives though from original OS. How can i mount them into Brunch plssss? 🙏🙏🙏

tejasraman, 2023-06-03:
@Nyta8591 it has a baytrail Celeron, that won't work with 101 >

Nyta8591, 2023-06-03:
> @Nyta8591 it has a baytrail Celeron, that won't work with >101. (Blame Google; they started to use ISAs that aren't supported on that CPU from >101)

Got it! Anyone has sub 100 version left? The oldest one of the website starts with 102…

also, will i be able to mount audio drives somehow plss? 🙏 could you please clarify what i should run in crosh?

tejasraman, 2023-06-03:
Yeah, I have a version 99 image..

Nyta8591, 2023-06-03:
> Yeah, I have a version 99 image... I'll send it once I go to my desktop

Thanks so much for all your help!

tejasraman, 2023-06-03:
@Nyta8591 you're welcome!  You also need to enable this flag (#consumer-auto-update-toggle-allowed) to see the option in settings to stop updates

Nyta8591, 2023-06-03:
> @Nyta8591 you're welcome! You also need to enable this flag (#consumer-auto-update-toggle-allowed) to see the option in settings to stop updates (otherwise your laptop will update past 100 and mess things up)

Awesome, will make sure to stop auto updates then. I really hope enabling audio flag for EOL chromebooks would be able to solve the issue, didnt work on the latest 111 though…

tejasraman, 2023-06-03:
@Nyta8591 you need to enable the chromebook audio on brunch options

tejasraman, 2023-06-03:
@Nyta8591 here is the image: [https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14526.69.0_rammus_recovery_stable-channel_mp-v2.bin.zip](https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14526.69.0_rammus_recovery_stable-channel_mp-v2.bin.zip) - it's an official rammus 100 image

Nyta8591, 2023-06-03:
> @Nyta8591 here is the image: https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14526.69.0_rammus_recovery_stable-channel_mp-v2.bin.zip - it's an official rammus 100 image

many thanks for this, will try it out soon

tejasraman, 2023-06-03:
You're welcome @Nyta8591

Nyta8591, 2023-06-03:
do i need to exactly match the Brunch version, for example if I will try 94 Chrome OS, Brunch release should also be 94 or can be later?

tejasraman, 2023-06-03:
I would recommend any Brunch version pre-103... 

This one should work: https://github.com/sebanc/brunch/releases/tag/r102-stable-20220625

Nyta8591, 2023-06-03:
Not sure if I should try 94 rammus or 100.... given my Chromebook's last update was v93 I think, safer to go with 94?

tejasraman, 2023-06-03:
No, go for 100... The AUE is arbitrary, they only dropped those ISAs later. 100 is a better release than 94 for a number of reasons

Nyta8591, 2023-06-03:
got it, ok, will try it soon!

Nyta8591, 2023-06-03:
> No, go for 100... The AUE is arbitrary, they only dropped those ISAs later. 100 is a better release than 94 for a number of reasons

Brunch 102+100 ChromeOS finally worked! I had to try different kernels to make audio work, and I only managed to get it working under Chromebook 4.4 kernel + “chromebook_audio” enabled under configurations. Bluetooth doesnt work though, but its minor… i guess it will be difficult to make it work?

tejasraman, 2023-06-03:
Try iwlwifi backport???

Nyta8591, 2023-06-03:
Hm, strangely i dont even see this option….
[image]

tejasraman, 2023-06-03:
The photo is not loading... I'll check once I get back, I'm at Ikea right now

Nyta8591, 2023-06-03:
[image]
sorry, here you go....

tejasraman, 2023-06-03:
What kernel is that??

Nyta8591, 2023-06-04:
I tried 5.1 with “chromebook_audio” enabled, but unfortunatelly it didnt fix the audio issue. Only the 4.4 chromebook kernel did

Nyta8591, 2023-06-04:
Let me try activating native image, maybe it will fix the audio…

tejasraman, 2023-06-04:
I said to enable `
native_chromebook_image
` too with 5.10

tejasraman, 2023-06-04:
Sorry, you sent that.when I was typing

Nyta8591, 2023-06-04:
Tried it, audio still not working, also tried on 5.4 kernel with same parameters you mentioned… thats a shame, i guess 4.4 chromebook is quite outdated (the only one that made my audio work, but only after debugging)

tejasraman, 2023-06-04:
I'll look into this a little further... can you boot into a Linux USB, run "lshw" and tell me the audio chipset??

Nyta8591, 2023-06-04:
If its not mentioned in any of the attached, I can do it tmrw only, since USB is in the room of someone who is asleep hahah 
[lspci, lsusb, lsmod.txt](https://github.com/sebanc/brunch/files/11644552/lspci.lsusb.lsmod.txt)
[byt-max98090.txt](https://github.com/sebanc/brunch/files/11644554/byt-max98090.txt)
[HiFi.txt](https://github.com/sebanc/brunch/files/11644555/HiFi.txt)

Many thanks for all your help! Wouldn't be able to make this machine work without you....

tejasraman, 2023-06-04:
There's a line missing at the top of the file for the maxim98090 codec

Add `Syntax 2` to the top of the UCM2 file inside the ChromeOS shell (not crosh)

Nyta8591, 2023-06-04:
Ok, will try tomorrow, going to sleep! Thanks so much once again

Nyta8591, 2023-06-04:
> There's a line missing at the top of the file for the maxim98090 codec
> 
> Add `Syntax 2` to the top of the UCM2 file inside the ChromeOS shell (not crosh)
> 
> And do this after **disabling native Chromebook image** (otherwise UCM2 won't apply)

Ok, tried googling, but still couldn’t figure out what you meant by CromeOS shell if its not Crosh… also not sure how to disable native chromebook image :(

Nyta8591, 2023-06-04:
also, looks some games from Playstore don't load, like Roblox/Need for Speed, apparently thats a ChromeOs limitation on non-native hardware.... I think it loaded fine on Bliss OS (which runs on Android 12), but sadly audio doesn't work there either... Bliss admin said if I can provide the source code devs can add drivers to the next build... would you know by chance where to get the source code for audio drivers? 

Thanks so much!

tejasraman, 2023-06-04:
> > There's a line missing at the top of the file for the maxim98090 codec
> 
> > 
> 
> > Add `Syntax 2` to the top of the UCM2 file inside the ChromeOS shell (not crosh)
> 
> > 
> 
> > And do this after **disabling native Chromebook image** (otherwise UCM2 won't apply)
> 
> 
> 
> Ok, tried googling, but still couldn’t figure out how to disable native chromebook image :( or add ‘Syntax 2’ to UCM2 file…

Type "shell" in crosh, and use nano (with sudo) to edit the UCM, and use alsaucm to load it

Go to brunch settings and redo without native chromebook image before doing this

tejasraman, 2023-06-04:
> also, looks some games from Playstore don't load, like Roblox/Need for Speed, apparently thats a ChromeOs limitation on non-native hardware.... I think it loaded fine on Bliss OS (which runs on Android 12), but sadly audio doesn't work there either... Bliss admin said if I can provide the source code devs can add drivers to the next build... would you know by chance how to get the source code for audio drivers? 
> 
> 
> 
> Thanks so much!

When you say this, do they crash or not show up in the play store?

Nyta8591, 2023-06-04:
> When you say this, do they crash or not show up in the play store?

so they download, but when I try opening the apps, they crash....

tejasraman, 2023-06-04:
Interesting… does this affect all apps?

Nyta8591, 2023-06-04:
> Interesting… does this affect all apps?

no, just some apps.... but I was told that its a ChromeOS limitation on non-native hardware...

tejasraman, 2023-06-04:
That shouldn’t be a thing… I’ll test the apps on a brunch system and tell you

Nyta8591, 2023-06-04:
Just checked on Bliss OS (Android12), Roblox loads into starting menu, but then also fails.. btw, dmesg in Bliss gives that, do you know if its also alsa config files issue or smth else?

[image]

Nyta8591, 2023-06-05:
> Type "shell" in crosh, and use nano (with sudo) to edit the UCM, and use alsaucm to load it
>
> Go to brunch settings and redo without native chromebook image before doing this

So i am using this guide to remount the original alsa configs i pulled from native ChromeOS, but I can't locate the correct alsa folder.... do you know how to locate it? Also, how to enter the developer mode, I am not sure if the system is rw to let me replace config files?  

https://askubuntu.com/questions/1189122/audio-chtmax98090-still-not-working-after-fix

[image]

Nyta8591, 2023-06-05:
Also, I would like to sideload APKs and by following this guide https://beebom.com/how-sideload-android-apps-chromebook/ and sadly I don't see any "Linux (Beta)" Option...  

[image]

tejasraman, 2023-06-05:
Also, you'd need to ask MrChromebox if you can't use virtualization...

tejasraman, 2023-06-05:
Tell him that you cant enable hardware assisted virtualization (on his GitHub issue tracker, that's more for these things)

tejasraman, 2023-06-05:
> So i am using this guide to remount the original alsa configs i pulled from native ChromeOS, but I can't locate the correct alsa folder.... do you know how to locate it? Also, how to enter the [...]

Use "nano /usr/share/alsa/ucm2/name_of_file.cfg" in shell to edit it.

Nyta8591, 2023-06-05:
> Use "nano /usr/share/alsa/ucm2/name_of_file.cfg" in shell to edit it.

ucm2 directory doesn't exist and ucm gives me an empty file? 

[image]
[image]

tejasraman, 2023-06-05:
It doesn't exist if native_chromebook_image is on. Turn it off and try after that.

tejasraman, 2023-06-05:
Check inside ucm2 and use sudo

Nyta8591, 2023-06-05:
> Check inside ucm2 and use sudo

There is nothing under ucm2, but with sudo i managed to alrer this one /usr/share/alsa/ucm/byt-max98090/byt-max98090.conf

do i enable native chromebook image now via  sudo edit-brunch-conf to check if it worked?

tejasraman, 2023-06-05:
Ok, something's wrong..... I need to check with these versions myself. I'll install chrome100 on a legacy laptop and tell you what to do to get audio

Nyta8591, 2023-06-05:
> Ok, something's wrong..... I need to check with these versions myself. I'll install chrome100 on a legacy laptop and tell you what to do to get audio

thanks so much for all your help, you are awesome!!!

Nyta8591, 2023-06-05:
> Also, you'd need to ask MrChromebox if you can't use virtualization...

https://github.com/MrChromebox/firmware/issues/474#issuecomment-1577008553

he replied....

[26 more replies]