sebanc/brunch #1674: The first setting stops at "Who's using this ChromeBook

thedanilfez, 2022-10-09:
Hello

When installing on a Lenovo G780 laptop, at the moment with logging into the account, it stays on "please wait" for a very long time, then for a second it gives an error related to wifi and then everything repeats
CPU: Intel Pentium 2020M
RAM: 8GB DDR3
VIDEO: intel hd graphics 4000 (IGPU) nvidia geforce gt 635m (dGPU)

Felzow47, 2022-10-10:
same for me

ameyjoshi3068, 2022-10-12:
Have you used chromros 103?

ameyjoshi3068, 2022-10-13:
Some devices have issues with latest chromeos. Try chromeos 94 and brunch 102 links are given

Rammus 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.74.0_rammus_recovery_stable-channel_mp-v2.bin.zip

Volteer 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_volteer_recovery_stable-channel_mp-v8.bin.zip

Grunt 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_grunt_recovery_stable-channel_mp-v6.bin.zip

Zork 94:
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14150.87.0_zork_recovery_stable-channel_mp-v7.bin.zip

Brunch 102 should be used with all ChromeOS releases before 103.
Brunch 102:
https://github.com/sebanc/brunch/releases/download/r102-stable-20220625/brunch_r102_stable_20220625.tar.gz

tejasraman, 2022-10-15:
Very similar to #1671
@Felzow47 could you try octopus 103 just for my own info?

tejasraman, 2022-10-16:
You know, your Celeron isn't compatible anymore but if it boots at all we can try 

Could you try coral and octopus recovery?

Felzow47, 2022-10-16:
> Very similar to #1671 @Felzow47 @thedanilfez could you try octopus 103 just for my own info? Strange that you and the person at #1671 have the same CPU.....

Sorry i can't try i have "flex os " installed on this computer sorry

tejasraman, 2022-10-16:
CrOS flex is a good solution. No android apps though. @Felzow47

Felzow47, 2022-10-16:
> CrOS flex is a good solution. No android apps though. @Felzow47

I hate chrome OS flex 
Bc you can't install nothing on it

tejasraman, 2022-10-16:
I understand :) As someone who uses Mac/Linux I totally get you. You could try it or try a Linux distro (way better if you want to use software on an old PC)
Sent with Spark

michalbaros, 2022-11-13:
I have same problem. R100 work for me, 
updating to r107 requires logging in again which ends in an endless wait

closed 2024-02-14 by thedanilfez

tejasraman, 2024-02-14:
PSA for anyone wondering: if you have a CPU older than Haswell/Baytrail DO NOT try to use Brunch, it won't work. Ivy Bridge works, but Android won't work  on Ivy Bridge.