sebanc/brunch #1678: Android apps and container crashes

bybenjamin, 2022-10-17:
Can you help with this situation?
Android apps crash and spin for a while and then reopen when I want to reopen. Attached I leave the dmesg output and screenshots
[dmesk.txt](https://github.com/sebanc/brunch/files/9800049/dmesk.txt)
[Screen recording 2022-10-17 14.20.45.webm](https://user-images.githubusercontent.com/91468765/196168859-702d5083-dbd1-4f48-87cd-83f541901c5b.webm)
@sebanc

ameyjoshi3068, 2022-10-18:
It is perfectly normal for android container to take time to start after boot. I dont know about the crashes though

bybenjamin, 2022-10-18:
Not during boot.  Apps crashing during use

Bwangster12, 2022-12-02:
I'm a lamen... but I found that the Android System Webview app cannot be on beta for me.  Apps worked, the app auto-updated and then apps stopped working.  I uninstalled the Android System Webview app, left the beta and the Android apps work fine again.