sebanc/brunch #1679: Brunch r105-unstable WiFi issues

tejasraman, 2022-10-17:
See #1677 
I recently installed Brunch on a Sandy Bridge MacBook Pro (kind of to prove a point - does @sebanc look at PRs in the main brunch repo? I'll make one about the sandy bridge... works w. caroline , will test coral and rammus) with CrOS 105 + Brunch 105 unstable. My WiFi card shows as "on", but no networks show up. I tried the blacklist in the wiki but it doesn't work.

Tejas

weberk, 2022-11-12:
Same here on a MacBook Air 11' mid 2013. Worked fine with r103 - not working after updating to r107 stable or unstable. Same diagnose as above: "My WiFi card shows as "on", but no networks show up."

tejasraman, 2022-11-12:
So the bcmwl driver is broken. What kernel are you using?

weberk, 2022-11-12:
I think the name of the driver module  is broadcom_wl. Here my kernel options: 
`options=broadcom_wl,enable_updates,acpi_power_button`

Since I am on r103 the driver is working now. The kernel is:
chronos@localhost /var/log $ uname -a
`Linux localhost 5.15.45-brunch-sebanc #1 SMP PREEMPT Thu Jul 21 18:12:24 UTC 2022 x86_64 Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz GenuineIntel GNU/Linux`

tejasraman, 2022-11-12:
I knew that and it works in a Linux env... Could you try a different kernel (eg 5.10-MacBook)?

weberk, 2022-11-12:
Dont know how to switch a kernel. 
Just to be clear: WIFI is working with my current kernel which was included in brunch r103.
Currently I am using ChromeOs r107 with Brunch r103. Everything operational now.

Whats not working is Brunch r107. Seems there was a regression between those two releases.

aguys123, 2022-11-13:
I have AzureWave AW-CE123H (BCM94352HMB) and have the same problem. As far as I remember, it's broken since Brunch r103 unstable 20220722 build. @sebanc

tejasraman, 2022-11-13:
Interesting..... I'll look into this.

sebanc, 2022-11-15:
This build should re-add broadcom_wl support:
https://github.com/sebanc/brunch-unstable/releases/tag/r107-unstable-20221115

However, note that the original chromeos version of wpa_supplicant is not compatible with broadcom_wl module and that this method will break at some point.

weberk, 2022-11-15:
Thanks for the fix sebanc and tejasraman! Wifi is working again in r107 unstable:
```
[   36.119301] wlan0: Broadcom BCM43a0 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
[   50.798625] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

```

closed 2026-01-08 by github-actions[bot]