sebanc/brunch #1681: [Realtek RTL8852AE WiFi 6 802.11ax PCIe Adapter] WIFI not working

rahman123456789, 2022-10-18:
I have installed the latest ChromeOS brunch file and it had installed successfully it. I come to try to begin the setup I can't see any network, I have tried to find a way to fix it from google but it didn't work. Please help me!

tejasraman, 2022-10-20:
There isn't a fix right now. I will work on it and get back to you. You need a kernel module :(

Tejas

tejasraman, 2022-10-20:
I'll try to compile tomorrow and send you a custom build to test. If it works, I'll submit upstream. I'll ask @sebanc about some stuff (kernel 5.7 is needed :( ) to see if this can work.

Tejas

tejasraman, 2022-10-21:
Oh did you try the 8852ae framework option in brunch-unstable?

RahmanAjeena, 2022-10-23:
it worked but now when i install and rebbot it says kerenl needed

tejasraman, 2022-10-23:
I need a pic of that.

RahmanAjeena, 2022-10-25:
> 

Its to late I have removed the os but ive tried reinstalling it always after a boot that brunch didnt boot from atleast 10 min it says needs kernel

loulou64490, 2023-01-11:
Hello @tejasraman 👋
you said you are working on a patch, is it working?
Thanks for all
Louis

tejasraman, 2023-01-11:
It should already be there, no patch needed. It's in the brunch options menu :)

closed 2023-02-08 by RahmanAjeena

sebanc, 2023-03-02:
Due to an issue, kernel 6.1 is not the default anymore in brunch-unstable builds (5.15 is the default) but you can select kernel 6.1 from the config menu