sebanc/brunch #1682: No sound on Microsoft Surface Laptop 4 (AMD)

A-johani, 2022-10-19:
Hi everyone ,

as mentioned in title line m i installed chrome os 105 with brunch 103 on SL4 (AMD) ryzen 5

everything is working great including the camera , boot , video BUT NOT AUDIO ,

i think the firmware is not included , any help is much appreciated .

tejasraman, 2022-10-20:
HI @A-johani could you try kernel 5.15? That's really new hardware but that should work on Linux 5.15.

Also why use chromeos? You lose your touchscreen but windows has support for it.

A-johani, 2022-10-20:
> HI @A-johani could you try kernel 5.15? That's really new hardware for 5.4 but that should work on Linux 5.15.
> 
> Also why use chromeos? You lose your touchscreen but windows has support for it.

Hi @tejasraman  i have tried 5.15 with the same result ,

also i don't like windows it either linux or chromeos and in both i loose the touchscreen which i don't mind .

tejasraman, 2022-10-20:
You could use Linux + Linux

tejasraman, 2022-10-20:
You could use Linux + Linux-Surface kernel for audio.

I'll see what I can do.

A-johani, 2022-10-20:
> You could use Linux + Linux-Surface kernel for audio.
> 
> I'll see what I can do.

thank you

A-johani, 2022-11-02:
Hi there ,

i tried the latest brunch r107 and chromeos 106 with the same issue , is there a specific flag i need to enable ?

A-johani, 2022-11-02:
BTW chrome os flex was fine with audio , but the keyboard & touchpad didn't work

tejasraman, 2022-11-02:
Probably just a driver that flex has.

PS: Please try Linux + Linux-Surface.

A-johani, 2022-11-02:
> Probably just a driver that flex has.
> 
> PS: Please try Linux + Linux-Surface. It'll work fine.

i have it on the same machine am dualbooting ,

but its not chrome os .

tejasraman, 2022-11-02:
I'll look into backporting the drivers from Linux surface.

If they're kernel modules it will be fine.

Tejas

mtimms1980, 2022-11-19:
Hi, wondering if you had any luck backporting drivers for audio? I am considering getting a Surface  Laptop 4 after I dropped my Matebook X Pro, I loved using ChromeOS on that laptop! 

Would touchscreen be possible?

tejasraman, 2022-11-19:
I'm still trying to do this... I need more info though. Can you run `lshw` in a Linux install? (SL3 audio works, not sl4 in ChromeOS)

closed 2026-01-08 by github-actions[bot]