sebanc/brunch #1685: RTL8761B Bluetooth not working

git448, 2022-10-22:
So the rtl8761b chipset is not working on brunch framework. however the (Linux) drivers for it are available (Links below) and it works on sideloading it on ubuntu(lib/firmware) or any other distro for that matter. So i was thinking is there any way i can sideload/install the driver in the brunch?

Method:-
https://medium.com/nerd-for-tech/how-to-install-unsupported-bluetooth-5-0-dongle-on-linux-4bf34aa99fed

Links:-
https://github.com/Realtek-OpenSource/android_hardware_realtek/tree/rtk1395/bt/rtkbt/Firmware/BT
https://github.com/Realtek-OpenSource/android_hardware_realtek/tree/rtk1395/bt/rtkbt/Firmware/BT/rtl8761b_config
https://github.com/Realtek-OpenSource/android_hardware_realtek/tree/rtk1395/bt/rtkbt/Firmware/BT/rtl8761b_fw

Thanks in Advanced.

sebanc, 2022-10-23:
Those firmwares are already present in brunch.

Could you post the output of "dmesg | grep hci" command in crosh shell ?

git448, 2022-10-25:
Yes. The issue was solved by changing the kernel to 5.15
Thanks!

closed 2022-10-25 by git448