sebanc/brunch #1690: Brunch Framework Bootloop

piemanny, 2022-10-24:
I'm not sure how to describe the error. Firstly I just want to say that Chrome OS was working on my PC. I had linux apps running and android apps running. Then the whole system crashed and when I tried to boot it again, it would show the splash screen for 2 seconds, and then restart again. 

I'm running a 5th gen i7 Dell laptop

ameyjoshi3068, 2022-10-25:
Have you updated chromeos recently? or do you have enable_updates option turned on? Can you tell which chromeOS and brunch version you were using?

piemanny, 2022-10-25:
I have just clean installed it again, which is a pain but there was nothing i could do

ameyjoshi3068, 2022-10-25:
Please close this issue if it is solved

piemanny, 2022-10-26:
It is not solved but I will close this issue though i think many others will discover this issue again

closed 2022-10-26 by piemanny