sebanc/brunch #1691: Crosh shell closing

git448, 2022-10-25:
Can anyone tell me how to open crosh ?
Ctrl+Alt+T is not working the shell just opens for a secong then closes down any help?

Brunch Version:103
Rammus Version:103

noidkn, 2023-05-26:
Такое же поведение, не мог установить chromeOS.

Получилось вот так;
Открываем терминал CTL+ALT+F2 , вводим имя пользователя chronos , пароль пустой.
Далее устанавливаем chromeOS командой sudo chromeos-install -dst /dev/disk , disk - это диск на который хотим установить, например /dev/sda . В моем случае ssd ноута.

signigelchan, 2024-01-20:
Same issue, I installed r119 `reef` in my laptop, and `CTL + ALT + F2`，then the window is opened, but if input any word, it's closing.

closed 2026-01-08 by github-actions[bot]