sebanc/brunch #1692: No Working essx8336

eloon-c, 2022-10-25:
Hello. The sound card is defined as snd_hda_intel, the sound does not work, although my sound card is essx8389, how do I fix it?

ameyjoshi3068, 2022-10-26:
have you tried disable_intel_hda option from chromeos settings?

eloon-c, 2022-10-26:
When selecting Brunch patches, it is not possible to select "disable_intel_hda", if you manually register "options=$options, disable_intel_hda" or "options=disable_intel_hda" then nothing changes. If you use the 5.10 kernel, then a mixer appears and the driver is now sof-skl.

My processor: 6th gemini lake.

eloon-c, 2022-10-26:
> have you tried disable_intel_hda option from chromeos settings?

When selecting Brunch patches, it is not possible to select "disable_intel_hda", if you manually register "options=$options, disable_intel_hda" or "options=disable_intel_hda" then nothing changes. If you use the 5.10 kernel, then a mixer appears and the driver is now sof-skl.

My processor: 6th gemini lake.

ameyjoshi3068, 2022-10-27:
> When selecting Brunch patches, it is not possible to select "disable_intel_hda", if you manually register "options=$options, disable_intel_hda" or "options=disable_intel_hda" then nothing changes. [...]

In brunch settings, You have to select framework options by 'space' and move to next screen by 'enter'

eloon-c, 2022-10-27:
> In brunch settings, You have to select framework options by 'space' and move to next screen by 'enter'

Item "disable_intel_hda" is missing

ameeno, 2023-09-18:
Hello. Did this ever get resolved?

I have no sound on brunch ChromeOS on a Tactus Geoflex 110. I believe this is the same issue.

pierro78, 2023-09-18:
> Hello. Did this ever get resolved?
> 
> I have no sound on brunch ChromeOS on a Tactus Geoflex 110. I believe this is the same issue.

I would not be surprised if the [essx8336](https://github.com/sebanc/brunch/issues/1692#top) issue has been solved in brunch with newer versions of the kernel as kernel 6.3 solved the issue on 2 different devices for me ... I need to check it out when I have time ...

ashah-apriori, 2023-10-16:
https://linux-hardware.org/?view=computers&vendor=Tactus&model=GeoFlex+110

cant get sound working, any help?

agrimpelhuber, 2023-11-27:
I think I have it running ... almost. And very simple.

I own a Tactus Geobook 140, with probably similar hardware. Most of all, essx8336 sound which hasn't been working so far. Just recently, I found this website / repository:

https://github.com/gnickm/sof-essx8336-debian-fix

I used some of the steps to get the sound running under a few linux distros, here's what I found regarding brunch. To do this, I installed the current version (118 rammus based) freshly on my Geobook 140. As Options, I only chose Kernel 6.1 and allow updates / pwa, nothing more. BIOS setting are all set to factory default.

Under Kernel 6.1, all the sof firmware files etc. seem to be already in place. As opposed to kernel 5.15 and below, the sound card now appears as a device in chromeos, but no sound.

Seems that the only missing thing now is the "setup-alsa.sh" to be run in chromeos. To do this, I need to:
* change it to "c 1" twice, because in the Geobook 140 the essx8336 is the second card after the USB webcam
* transfer the script to a partition without "noexec" flag (I use /root)
* run the script with sudo from the shell (ALT+CRTL+F2, login "chronos")

After that, the sound card works, both with speakers and headphones. Here's my version of the script
```
#!/bin/bash

echo "--- Setting up ALSA output for ESSX8336"

# This may be different for your system, you can check it by running aplay -l
OUTPUT_DEVICE='-c 1'

amixer $OUTPUT_DEVICE cset name='Speaker Switch' on
amixer $OUTPUT_DEVICE cset name='Headphone Playback Volume' 3,3
amixer $OUTPUT_DEVICE cset name='Right Headphone Mixer Right DAC Switch' on
amixer $OUTPUT_DEVICE cset name='Left Headphone Mixer Left DAC Switch' on
amixer $OUTPUT_DEVICE cset name='DAC Playback Volume' 999,999
amixer $OUTPUT_DEVICE cset name='Headphone Mixer Volume' 999,999
amixer $OUTPUT_DEVICE sset Headphone 3

echo "--- Setting up ALSA output for ESSX8336"

# This may be different for your system, you can check it by running arecord -l
INPUT_DEVICE='-c 1'

amixer $INPUT_DEVICE cset name='ADC PGA Gain Volume' 7
amixer $INPUT_DEVICE cset name='ADC Capture Volume' 150
amixer $INPUT_DEVICE cset name='Internal Mic Switch' on
amixer $INPUT_DEVICE cset name='Headset Mic Switch' on
amixer $INPUT_DEVICE cset name='Differential Mux' 'lin2-rin2'

echo "=== Done!"
exit 0

```
Only snag: I need to run this after every boot.

@sebanc , can this be turned into a permanent feature / patch for the essx8336 sound card? Everything else (sof firmware, kernel > 5.17) seems to be there in the latest brunch OS.

ameeno, 2024-02-16:
Going to try again this weekend using your guide @agrimpelhuber , 
The latest brunch should be using a newer kernel, which may work OOB?

pierro78, 2024-02-21:
> Haven't checked any newer kernel than 6.1 on brunch yet (is there a newer one?), 
@agrimpelhuber :  kernel chromebook-6.6 added in https://github.com/sebanc/brunch-unstable/releases since Brunch r120 unstable 20240201 ...

@ameeno did you have success on your device ? I think I am going to try on my n4020 Kuu Yobook M ... google chrome working quite well on it on Fedora LXDE but I miss the "top url toolbar autohide" feature of chromeos (yandex not working as fast as chrome on Fedora LXDE ...)

pierro78, 2024-02-21:
ok , tried with kernel 6.6 on latest brunch unstable but no sound on my device ... I am going to try to write a setup-alsa.sh like agrimpelhuber did ... ( I ll try to check how Fedora with kernel 6.7 has this working ...)

pierro78, 2024-02-21:
OK the **setup-alsa.sh** given by agrimpelhuber above ( https://github.com/sebanc/brunch/issues/1692#issuecomment-1828425632 ) works nicely on my **Kuu Yobook M** **if I set** 
**OUTPUT_DEVICE='-c 0'**
in it ... 

... still a little pain to have to run it manually after each boot though but I can now enjoy watching and listening nicely to youtube videos on my Yobook M with its great screen !

agrimpelhuber, 2024-03-01:
@pierro78, I just discovered that on teh very latest brunch version (chromeos Version 121.0.6167.212 updated the normal way, and Brunch r121 stable 20240224), there is a fix in the brunch settings (actually 2) to run the script on bootup. One for card 0, one for card 1. Works great on 2 different Geobooks-140. One is permanently connected to HDMI (so the internal sound is card 1), one is standalone (internal sound card 0). Both work fine with the respective script - yeah. Beith with Kernel 6.6, but also might work with 6.1 as before.

I haven't yet tested whether you can run both scripts, in case the internal sound changes from being card 0 to card 1 if HDMI is connected on startup. Will do that right away. But in the meantime, THANKS @sebanc and every one else involved for fixing this.

pierro78, 2024-03-01:
> @pierro78, I just discovered that on teh very latest brunch version (chromeos Version 121.0.6167.212 updated the normal way, and Brunch r121 stable 20240224), there is a fix in the brunch settings [...]

thanks so much for the tip !! ... just did the upgrade to Brunch r121 stable 20240224 from the pwa, selected essx8336 on card0 in the settings and it s working great on my Kuu Yobook M !!! ... thanks again !! and thanks also for the awesome work by sebanc !

pierro78, 2024-03-01:
> > @pierro78, I just discovered that on teh very latest brunch version (chromeos Version 121.0.6167.212 updated the normal way, and Brunch r121 stable 20240224), there is a fix in the brunch settings (actually 2) to run the script on bootup. One for card 0, one for card 1. Works great on 2 different Geobooks-140. One is permanently connected to HDMI (so the internal sound is card 1), one is standalone (internal sound card 0). Both work fine with the respective script - yeah. Beith with Kernel 6.6, but also might work with 6.1 as before.
> > I haven't yet tested whether you can run both scripts, in case the internal sound changes from being card 0 to card 1 if HDMI is connected on startup. Will do that right away. But in the meantime, THANKS @sebanc and every one else involved for fixing this.
> > Edit: Just tested running both scripts on both machines, didn't work.
> 
> Hi, Sorry, Dont understand, How do I get the geobook-140 soundcards to work properly with latest brunch r121 stable? please confirm the steps :D

@ameeno : at boot time choose "brunch settings", then in second settings screen choose "essx8336 on card1"

agrimpelhuber, 2024-03-02:
@ameeno, you might also need to select "card 0" instead if card1 does not work. I have 2 Geobooks-140, and they behave differently. Might have something to to with the fact that I broke the screen on one of them, and use it permanently on a HDMI display now.

Important: You need to use a kernel >6. 6.1 should work, but I'm using 6.6.

Kourosava, 2025-04-10:
Hi!
I have a Huawei Matebook d15 with ESSX8336 and audio isnot working. I am new to all this and i dont know what to do. Can someone please guide me?

agrimpelhuber, 2025-04-12:
@Kourosava, did you read and follow the above before opening a second, new issue?

Kourosava, 2025-04-14:
> [@Kourosava](https://github.com/Kourosava), did you read and follow the above before opening a second, new issue?
sorry for that!

I want to install sof-bin-2025.01.1 but i do not know how to do it. Can someone help me please?

Kourosava, 2025-04-16:
> * transfer the script to a partition without "noexec" flag (I use /root)
> * run the script with sudo from the shell (ALT+CRTL+F2, login "chronos")

can someone please guide me on those 2 commands? How to transfer and how to run the script

agrimpelhuber, 2025-04-16:
Hi, sorry for the late answer. I can't properly recall the history, but as far as I remember, that might be not necessary.

I seem to remember at some point in the past, the kernel and the associated sof firmware in brunch were sufficiently up to date for the essx8336.

So here the question: did you do the following things?

- in the brunch settings, choose the newest kernel
- also in the brunch settings on the second screen, choose either the "essx on card 0" or the "essx on card 1" option - just try it

That might do the trick, it's worth a try. If it doesn't, try different kernels and the 0/1 setting

Good luck

Kourosava, 2025-04-16:
I tried 6.1, 6.6 and 6.16 kernels. Still no sound

agrimpelhuber, 2025-04-27:
@Kourosava sorry for the late reply. And sorry for not being able to help you any more: I went through a lot of the past postings, and here is what I get from them:

- The main thing I did was figure out the instructions in [this comment](https://github.com/sebanc/brunch/issues/1692#issuecomment-1828425632) from my experiments with generic linuxes.
- As I wrote there, I was lucky when all the necessary files (kernel drivers and compatible sof firmware) started to appear in the kernels / chrome OSses / brunch versions, Therefore, I just had to ask sebanc nicely to run the alsa-setup script on startup, and make it configurable on the kernel / boot options of brunch (card 0/1)
- I have to stress again that - opposite of what I did with generic linuxes for that hardware - I did nothing to add kernel modules or sof firmware to chromeos / brunch

The only chance I see for your hardware:

- Try to get the alsa-setup script running manually on your hardware, as described above. If that fails, it might have a number of reasons (different hardware from the Geobook 140s I use, wrong kernel or sof firmware ...) ... or they have changed the way the shell works, so that you can't run it manually at all.
- If  you are lucky, it might be a minor thing in the script. For example, a minor change in the script might be needed. I could easily imagine that apart for cards 0 and 1, your hardware might be card 2 or 3. That might be fixable by sebanc.

Apart from that, it might not be worth the trouble.

Good luck!

ameeno, 2025-10-20:
Any chance we could get this to work on ChromeOS flex? I tried that last night, and no sound.

I was thinking that if I could get a start up script to enable sound on the flex it might work better for my five year old?