sebanc/brunch #1701: ChromeOS Flex Support

TerminalPR0, 2022-11-08:
chromeos flex can be patched with brunch?

stxrrynight008, 2022-11-13:
why would it need to be?

TerminalPR0, 2022-11-13:
it would be nice, since this version of the OS does not require
virtualization, and partially works on MBR disks (it works fine on a flash
drive, but after installation it does not see the installed system)

JimDude7404, 2022-11-26:
> and partially works on MBR disks (it works fine on a flash drive, but after installation it does not see the installed system) вс, 13 нояб. 2022 г., 23:43 stxrrynight008 ***@***.***>:
You could just instead of burning to a USB try burning to a partition on your drive instead. It would work better. Then it wouldnt show as installed but hey! you got it on a MBR drive

TerminalPR0, 2022-11-26:
the problem is that I not able to install multiple drives in my laptop, so
I can't do that(

x-0D, 2022-11-26:
It would be good solution i think - ChromeOS features with ChromeOS Flex hardware support

ccchan234, 2023-02-28:
will it be done?

os flex really hate to be duat boot-ed,
others have tried a lot of methods.

would be nice if brunch could make it.

thanks

Bruh124567, 2023-02-28:
Yes ok but then just dualboot normal chrome os because it would make no sense to do the same steps to get a more dumbed down chrome os

ccchan234, 2023-02-28:
> Yes ok but then just dualboot normal chrome os because it would make no sense to do the same steps to get a more dumbed down chrome os

if brunch+chromeos could get updates rock solid i'll love to use it.
but what i heard is not.

ps: i never personally installed brunch+chromeos.

i installed os flex, it is signed, from google, i trust it.

Bruh124567, 2023-03-01:
It Will also be unstable because of the way brunch sets up the partitionsFeladó: ccchan234Elküldve: 2023. február 28., kedd 11:50Címzett: sebanc/brunchMásolatot kap: Bruh124567; CommentTárgy: Re: [sebanc/brunch] ChromeOS Flex Support (Issue #1701) Yes ok but then just dualboot normal chrome os because it would make no sense to do the same steps to get a more dumbed down chrome osif brunch+chromeos could get updates rock solid i'll love to use it.but what i heard is not.ps: i never personally installed brunch+chromeos.i installed os flex, it is signed, from google, i trust it.

Bruh124567, 2023-03-01:
sry i replied with email:  It Will also be unstable because of the way brunch sets up the partitions

P.S: Normal Chrome OS is also signed by google and usually you download it from google's update server.