sebanc/brunch #1718: Chrome OS Consumes Too Many Resources

bybenjamin, 2022-11-24:
I installed chrome os and then fyde os on laptop with 4gb ram. After activating Android, Chrome Os will have 400mb idle ram and Fyde Os 1.5mb ram. For this reason, there are contractions in Chrome Os android applications. I tried different images. No problems other than contractions. Is the problem in the framework of brunch?

JimDude7404, 2022-11-26:
This isnt a brunch issue. Nor is it even a real issue. I am on my laptop with 4GB of RAM and it has over 1.5GB free with the Google Play Store enabled. I think there is something wrong with either the recovery you are using or smth. Brunch is just a FRAMEWORK not the actual OS itself. If you want a full Chromebook without issues your gonna need to buy one. Also this post shows very little issues. I'd reccomend that you just close this issue...

bybenjamin, 2022-11-26:
> This isnt a brunch issue. Nor is it even a real issue. I am on my laptop with 4GB of RAM and it has over 1.5GB free with the Google Play Store enabled. I think there is something wrong with either [...]

Calm down dude I already have a full chromebook I prefer it for brunch burrito

bybenjamin, 2022-12-03:
Google switched to arcvm with android 11. but it consumes a lot of resources and causes the battery to die prematurely. Is there a recovery file running android 9 for the Jasper Like platform?

JimDude7404, 2022-12-03:
You can switch to Coral

kocoman1, 2022-12-17:
> 

what it mean