sebanc/brunch #1744: ELAN fingerprint drivers

1457384613gh, 2022-12-25:
ELAN fingerprint drivers.
Do they work?

imatefx, 2023-01-17:
I just installed chromeOS on my Asus Flow X13 its ELAN fingerprint is not detected, but it shows up in Manage usb devices in Linux development settings

tejasraman, 2023-03-04:
ChromeOS has _NO_ Elan fingerprint support whatsoever @1457384613gh @imatefx

tejasraman, 2023-03-11:
> OK, we are asking too much!
> By the way, another question.
> Turning .img file into a lvm partition, could it be installed into a lvm partition?

Never really experimented with lvm, sorry :)

tejasraman, 2023-03-11:
@1457384613gh I do, but LVM was never really a practical solution for me.

tejasraman, 2023-03-17:
I think it can if you have windows installed as mbr.

tejasraman, 2023-03-17:
Ohhhhhhhh... No, it won't do that.
You might be able to do this using clover but that's the extent of my knowledge on the situation

tejasraman, 2023-03-17:
Refind isn't clover.....
You need the clover loader to load an efi file for you

tejasraman, 2023-03-18:
> Tell you a good news.
> I've tried.
> BIOS->MBR->PBR->DUET->refind_x64.efi
> It can be loaded without Clover.

Ok so duet loader(from opencore I assume) worked :)

tejasraman, 2023-03-18:
What about Brunch's mbr patch?

Also, Duet seems like a good way to emulate EFI on BIOS systems. Thanks, I never really thought of duet as a solution for anything other than opencore (and I don't have any old bios systems anymore, my oldest is a 2011 Mac and a 2013 (8.1-era) PC)

tejasraman, 2023-03-18:
Ok so your laptop has no efi support?

And I do understand that the mbr patch has no multi boot support……

tejasraman, 2023-03-18:
1st gen hasn’t been supported since ChromeOS 81.

tejasraman, 2023-03-18:
[image]

tejasraman, 2023-03-18:
Sorry, I'm confused. M450 and 45xx series are 2 different things :)

tejasraman, 2023-03-18:
And there is a GPU ROM patch to add UEFT Graphics Output Protocol (GOP) support to GPUs, but it's only for dedicated NVidia/AMD cards.

tejasraman, 2023-03-18:
Have you tried logging in to google? It will fail, I’ve tried it with 2nd gen (and it had a UEFi GPU because of Apple too) and it fails.

tejasraman, 2023-03-18:
If you check a `dmesg` after logging in you’ll see a lot of “invalid CPU instruction” messages.

reven doesn’t rely on those ISAs but it has no Play Store either.
I guess fyde is your best option :)

tejasraman, 2023-03-19:
ChromeOS (Settings) is not part of the grub; rather, it can even be run from inside of chromeOS `sudo edit-brunch-config`

And btw, I'm not the Brunch developer, @sebanc is.

tejasraman, 2023-03-19:
Ok yes I am :)
I'll look into your proposals but "contributor" means I've submitted a few PRs :)
I'm making a grub theme to test for future implementations, but you need to ask someone else about ventoy :). (I love ventoy but I know nothing about it beyond the basics despite 5out of 6 of my boot USBs using it)

tejasraman, 2023-03-19:
And the bootsplashes are all in .png format and at static _960x540_ (shocking, I know) and are surrounded by (black) borders on higher resolution displays (e.g 1920x1080 - try a light theme and see for yourself). 

I can see the value of vector bootsplashes that are truly scalable (try booting brunch in 4K and it'll be so small) but I would need to look into it more.

Also, a progress bar for the bootsplashes certainly isn't impossible (at least for rootFS where there is a percentage in verbose mode)

tejasraman, 2023-03-19:
Ok here's what I kno
[image]
- Brunch is using FBV to display generic images on screen.
- Brunch is manually appending logs to kmsg

tejasraman, 2023-03-19:
> Could FBV support to show .SVG?
No, FBV is old and cannot do that. At this point an alternative needs to be found if you want to have SVG boot splashes.

tejasraman, 2023-03-19:
Ok thanks @1457384613gh I didn't know that :)

But the exciting part is that there is ext4 support (and brunch performs best with ext4, not NTFS)

tejasraman, 2023-03-19:
@1457384613gh I know that Ventoy boots internal grub. I'm not sure if brunch images have an internal ESP.

tejasraman, 2023-03-19:
Ok thanks so much @1457384613gh

I'll test some things and try to make it work as best I can..

tejasraman, 2023-03-19:
Because I'm not "in contact" with every Brunch contributor; contributions are usually done individually.

tejasraman, 2023-03-19:
And I’m not the person to ask about Ventoy either… I’ll try though

tejasraman, 2023-03-19:
@1457384613gh For vectorized (non-PNG) bootsplashes I learned about [jfbview](https://github.com/jichu4n/jfbview), a PDF viewer for framebuffers. This will allow high resolution display of bootsplashes

tejasraman, 2023-03-20:
Yes, my ID for element/matrix is traman124:matrix.org

tejasraman, 2023-06-01:
That looks like a verbatim copy of brunch’s grub2…..

tejasraman, 2023-06-01:
Also, there is no way to use ttfs and pngs (like in Plymouth) with a simple ttf tool. The closest that I could get was jfbview (which shows hi res bootsplashes) (see my fork of brunch-unstable)

tejasraman, 2023-06-02:
I see no point in including a txt file system in brunch; people aren't supposed to edit grub.cfg. I don't foresee a large number of people using non stock locations for the img files...

Can you explain the potential usage for this???

closed 2026-01-05 by github-actions[bot]