sebanc/brunch #1749: ES8336 broken audio

erenmete, 2023-01-04:
Support in Linux kernels for this audio card is fairly new and the mainline kernel 6.1 and above has commits that get it working. Can these commits be cherry picked and implemented? 

[image]

[image]

mirek190, 2023-01-25:
yep ... for me with that audio es8336 not working either ...

tejasraman, 2023-03-04:
Hi @elfametesar @mirek190 6.1 is now in the unstable builds!

mirek190, 2023-03-04:
> Hi @elfametesar @mirek190 6.1 is now in the unstable builds!

Yea ..but kernel 6.1 ans sof drivers requires new sof binaries 2.0+ and ucm .

I actually made my own driver for kernel 5 15 and is compatible with old sof binaries.

tejasraman, 2023-03-04:
I know… see my attempts to get steamdeck audio on ChromeOS. A lot of outdated components which makes something as simple as UCMs impossible.

But seriously, submit a PR to Brunch Unstable. We’d love to have that driver available in Brunch.

mirek190, 2023-03-06:
My driver is for intel sof 8336. 
AMD has a different one.

tejasraman, 2023-03-06:
@mirek190 I know, just saying about a similar issue.

closed 2023-03-24 by elfametesar

mayaru, 2023-06-10:
Hi @mirek190 
I have the same problem with the intel sound card. any chance you can share your driver so i can test it out?
Thanks a lot

ashah-apriori, 2023-10-16:
Not working for me too, and manjaro works with sound. not sure how to make it work with brunch. i tried building a kernel and I also tried adding sof drivers. but not working yet.