sebanc/brunch #1762: can't launch Google play store

Ryan15335, 2023-01-18:
Hi I can't open the google play store because when it does it asks me to accept the license and it configures the google play store but it fails with error code 7 or never ends the configuration. Config if needed : Grunt AMD A6-9225

benjamin920101, 2023-01-20:
I can't open the google play store in ChromeOS 109

lmore377, 2023-01-21:
Just use and stay on ChromeOS 108 for now. It seems to be the most reliable for me

BlueOutside, 2023-01-29:
I installed the latest brunch-unstable build and it didn't worked i had the same error code 😔

Pumpino, 2023-03-01:
I just did a fresh install of Brunch unstable (28/2/23) with ChromeOS 110 after having had a break from ChromeOS for a few weeks. I decided to have a go at using Android apps, whereas I previously only used linux apps.

I've found that I too get error code 7 when trying to load the play store. To be honest, I'm disappointed that this has apparently been present for six weeks across both v109 and v110. It's a significant issue for people that rely on certain apps.

samu04072013, 2025-02-18:
I was tring out brunch on a Lenovo Ideapad S145-14AST 81ST and I also get error code 7 on play store and could not connect using ADB. I followed the web doc instructions during install but I'm in stoney bridge not on ryzen so i used grunt instead of guboz as gumboz didn't boot after brunch only worked wit a video guide using chromeOS 115 and zork board but it said system used 1022 GB of disk and I have 1TB so I could not update to chromeOS 135

ZagoskinBeatz, 2025-03-19:
I install ChromeOS and hoped to use android applications, but also first received error 8, and then error 7. Used different versions Brunch and different versions of the system. Even the account used and even VPN, but nothing helped. My laptop - HP 15-rb012ur 15.6