sebanc/brunch #1766: System partition taking 59.8 Gb

Bubbles78959, 2023-01-25:
So I installed brunch on a external drive, and the chrome system partition is taking up most of the space. A little amount can actually be used by files. I even tried the resize-data command and it did nothing. What can I do to fix this?
[image]

knguyen1401, 2023-01-27:
Yeah I'm also experiencing this. I'm seeing my "System" storage is increasing lately.

Bubbles78959, 2023-01-29:
Oh I forgot to add, the hard drive is 60 GB, so the amount the system takes makes no sense with how much is allocated for other files-

Bubbles78959, 2023-01-29:
So I just installed it on another hard drive which is 1 TB, and the system takes 153 GB... Is it just hogging free space or is that just how much chrome uses?
[image]

knguyen1401, 2023-01-30:
Can you keep track of the system storage size?

Cateners, 2023-07-02:
I have the same issue. I installed brunch on a 512GB hard drive, and the system shows that it occupies about 509GB of space in the settings. However, when I switched to the Linux system to check the disk status, I found that there was approximately 470GB of unallocated space. So, I directly appended it to the last partition. Now, the settings show that 77GB of space has been used. Although it's still a bit much, the remaining space is sufficient.