sebanc/brunch #1769: My Bluetooth is completely dead and not even showing the option.

Anuragisme, 2023-01-29:
@tejasraman Bluetooth in my Chrome OS is completely gone. In the morning it was working well. I tried to charge a phone via usb suddenly my Chrome OS shut down and since then Bluetooth is not working.

tejasraman, 2023-01-29:
Have you tried manually updating brunch to the same version? Also, what is the output of `lspci | grep Wireless` (if you have a wifi/BT combo card like most laptops)

Anuragisme, 2023-01-30:
I have reinstalled Chrome OS thrice but the Bluetooth option is completely dead. 

But where to put this command.

Anuragisme, 2023-01-30:
Here is the screenshot
(https://user-images.githubusercontent.com/99806758/215455044-f71ee864-dac2-44ab-ade1-dd2173d20953.png)

Anuragisme, 2023-01-30:
Here the output of the command 
[image]

tejasraman, 2023-02-01:
Enable 8821ce in the menu with the PWA option.

Anuragisme, 2023-02-01:
@tejasraman , BUt from where to enable this. I don't know how can I do that?

tejasraman, 2023-02-01:
Go to crosh (Ctrl+Alt+T), type `shell`, and then type `sudo edit-brunch-config`. Enable 8821ce as well as any other options (such as PWA) and reboot. Remember to use kernel 5.10 (recommended) or 5.15.

Tejas

closed 2023-02-04 by Anuragisme

tejasraman, 2023-02-04:
@Anuragisme is this resolved?

Anuragisme, 2023-02-05:
@tejasraman Actually the problem was in my hardware.So, I got it repaired
in the service centre.

Thank You for your support and help, I would be very grateful to you.

tejasraman, 2023-02-05:
You're welcome @Anuragisme