sebanc/brunch #1770: Dell latitude 5480 not detecting charging (always reporting discharging)

tungmeister, 2023-01-29:
Hi,
I'm having an issue with my latitude 5480 in that the battery state is always reporting as discharging, even when the charger is connected. The laptop charges fine but the state never changes from discharging. I didn't use to have this issue but I stopped using brunch for a while and now after having come back to it and updating this issue has presented.

tungmeister, 2023-04-09:
@sebanc any suggestions?

giaur500, 2024-04-21:
Any updates? That issue is actually serious. For example, I am not able to update Linux version, because ChromeOS asks to connect charger, even when charger is connected. So, it's impossible to update Linux VM

tungmeister, 2024-04-22:
Not that I'm aware of, I tried what was suggested in that referenced issue but it didn't help.

giaur500, 2024-10-07:
In my case issue is fixed now. Connected charger is detected correctly. Chomeos/Rammus, latest stable brunch version

closed 2026-01-01 by github-actions[bot]