sebanc/brunch #1771: Only 6GB RAM on 8GB RAM

NokiPlays, 2023-01-31:
Hi! Today i installed ChromeOS on my Laptop with Intel Core i3 11th Gen using Linux (Ubuntu) I booted into the OS and everything worked fine, no lag all smooth but when i tried to play Minecraft Bedrock Edition, it was unplayable, so i wanted to see what was doing this, and i saw there was only 6gb RAM while i have 8GB RAM and there was only 300mb left, all i have is Adblocker extension and Minecraft from PlayStore, and Messenger. Please help me 😊

lentas1, 2023-03-18:
and its GB not gb

tejasraman, 2023-03-18:
@lentas1 GB is the same as gb. GiB (gibibyte) is different than GB(gigabyte). Lowercase vs uppercase means **nothing**

lentas1, 2023-03-19:
ur confused

lentas1, 2023-03-19:
https://www.mixvoip.com/what-is-the-difference-between-gb-and-gb

lentas1, 2023-03-19:
They sound similar, and this causes a lot of confusion among users who are not tech-savvy. Sometimes hosting companies use the terms and the abbreviations interchangeably when they try to sell their storage services, which is not a good practice and customer-friendly. That way, you can be misled about the size of the storage space you pay. If in the hosting plan they have used gigabits as measuring units and you do not pay attention, you will be unpleasantly surprised to find out that you have eight times smaller storage capacity than expected.

tejasraman, 2023-03-19:
@lentas1 I know the difference between a gigabyte and a gigabit (gigabyte is 8x a gigabit) but when we’re talking about RAM, gigabytes are implied. 

Gb vs GB is different. gb is the same as GB.

Idk who decided that we should have GB, gb, and GiB.

There is no need to complain to someone who has a ChromeOS related problem about storage standards. Please don’t be so pedantic

lentas1, 2023-03-19:
ok

tejasraman, 2023-03-19:
I'm really sorry @lentas1