sebanc/brunch #1774: Brunch 109 - Can not start Linux

dazbeattie, 2023-02-02:
Using a Surface laptop 3. just put a fresh install of Brunch 109 with Volteer 109. I have PWA and updates enabled. Kernel is 5.15
Machine is great. Can install the linux container and it started after the install.

After a reboot, the linux container will not start. Get the following error :

[==\        ] Starting the virtual machine Error starting crostini for terminal: Launching vmshell failed: Error starting crostini for terminal: 54 (START_TERMINA_VM_TIMED_OUT)

If I uninstall linux and re-enable it, Linux starts after the install, but fails after a reboot every time.

I also have a Surface Pro 6, but still on 107 and kernel 5.15, no issues. Do not know if it a Surface Laptop or a 109 issue.

Thanks for great tool and your dedication to all of us.

githubston, 2023-09-10:
Many of the failed installations of cros-termina are related to a security problem, hardware incompatibility... and servers... penguin uses iPv4, iPv6... A quick solution to partially solve the error : 54 for example: may be to: set the DNS of (Google) to 8.8.8.8 is 1.1.1.1 (Cloudflare) if you don't have an iPv6...

dazbeattie, 2023-12-08:
Closing issue as not relevant to later Brunch and ChromeOS versions

closed 2023-12-08 by dazbeattie