sebanc/brunch #1782: Advise wanted: Why dont you people use fydeos from China?

ccchan234, 2023-02-08:
i got a celeron win10 tabet, fxxking slow. but i need to use .exe;

yet for web browsing i wish fast, chrome + some android apps.

i need dual boot.

i use chinese chuwi hi10x (not hi10 go), the one with intel/realtek sound card.

i could dual boot win10+ bitlocker /w fydeos, a chromium + android environment.

active stylus supported.

i never tried brunch, 
need too much install steps.
got update problem.

the only one thing: mostly from google + foss, so not closed source as fydeos,
which is from China.... (i am chinese..)

4stax, 2023-02-19:
This isn't a issue.

bybenjamin, 2023-03-01:
brunch fyde os ile çalışıyor.  ama android alt sistemi etkinleştirmiyor

closed 2026-01-01 by github-actions[bot]