sebanc/brunch #1785: Stuck on booting Kernel, then restarts. AMD processor and using zork image.

bruh-person, 2023-02-08:


bruh-person, 2023-02-09:
INFO: MBR LEGACY BIOS SYSTEM, 64B SYSTEM

tejasraman, 2023-03-04:
Ok, @bruh-person what's your CPU? Just "an AMD processor" doesn't suffice

Priyanshu6517, 2023-03-04:
Same issue I have an amd a6 4400m cpu

bruh-person, 2023-03-04:
I use legacy BIOS and I will get the processor now

> Message ID: ***@***.***>
>

bruh-person, 2023-03-04:
Also, you are very late.

tejasraman, 2023-03-04:
@Priyanshu6517 An A6-4400M won’t work with brunch (it’s based on the Piledriver architecture from the Trinity series. The only AMD that will work are a very small subset of newer A-series CPUs and all Ryzen CPUs)

tejasraman, 2023-03-04:
@bruh-person what do you mean “late”?

bruh-person, 2023-03-04:
> @bruh-person what do you mean “late”? And what’s your CPU?

bruh i posted 3 weeks ago and you come now

bruh-person, 2023-03-04:
I'm getting it.

bruh-person, 2023-03-04:
apparently it is a AMD Turion 64 X2 Mobile Technology TL-60 (2) @ 800MHz

bruh-person, 2023-03-04:
what i got from neofetch in partedmagic
:\

tejasraman, 2023-03-04:
@bruh-person thats incompatible, maybe try chromeos flex?

bruh-person, 2023-03-04:
the pc i'm using right now is **Ryzen**

bruh-person, 2023-03-04:
so i should just use a random image or something until it works :+1:
@bruh-person

tejasraman, 2023-03-04:
Is this a joke? (AMD Turion 64 X2 Mobile Technology TL-60 (2) @ 800MHz)

bruh-person, 2023-03-04:
sure if you want it to be

tejasraman, 2023-03-04:
Ok what computer are you trying to install ChromeOS on?

bruh-person, 2023-03-04:
laptop

bruh-person, 2023-03-04:
I'm using my desktop pc to github issues

tejasraman, 2023-03-04:
Oh, the laptop has the turion? Ok so you can't install it on the laptop (the CPU is from 2007. Could you try ChromeOS flex?) but it would work with zork on the desktop.

bruh-person, 2023-03-04:
ChromeOS Flex doesn't support Google Play Store apps: https://support.google.com/chromeosflex/answer/11542901?hl=en#:~:text=Supported%20VMs%20and%20apps,Android%20apps%20or%20Google%20Play.

bruh-person, 2023-03-04:
and i'm installing it on a usb

tejasraman, 2023-03-04:
That's true, but Brunch doesn't support your CPU and Flex will...

Also, the certified models for flex don't matter - every PC I've tested works with it, even really old stuff (2008, 2011)

bruh-person, 2023-03-04:
also i could use a vm on the desktop, since vmware just replicates host cpu and stuff :\ but why wouldn't brunch support VMs?

tejasraman, 2023-03-04:
Because v,ware virtualizes your gpu.

bruh-person, 2023-03-04:
QEMU?

tejasraman, 2023-03-04:
Not unless you want unaccelerated... and IME it doesn't work at all in qemu

tejasraman, 2023-03-04:
(see this image:)
[image]

bruh-person, 2023-03-04:
> (see this image:) [image]

ya ok

tejasraman, 2023-03-04:
So do you understand that you cannot install brunch on your laptop?

(Try fydeos as well, it has Android support but isn’t ChromeOS per se)

bruh-person, 2023-03-04:
y

bruh-person, 2023-03-04:
ok

closed 2023-03-04 by bruh-person