sebanc/brunch #1786: Rebootloop after installing ChromeOS

ChunkyZacko5, 2023-02-11:
Hello, I was using ChromeOS, I clicked on ChromeOS entry and I got reboot when screen goes "Starting Brunch Framework...".

v-mod, 2023-02-26:
I have the same issue. Have you found a solution yet?

v-mod, 2023-02-26:
I have tried reinstalling. It worked fine on the first boot.

tejasraman, 2023-03-04:
This usually means there's a hardware incompatibility. Please tell me your computer model (CPU too please) @ChunkyZacko5

Priyanshu6517, 2023-03-04:
Im also facing the same problem I have an amd a6 4400m CPU and my PC does have UEFI support

v-mod, 2023-03-04:
I have just reverted back to Brunch 107 and now it works fine. Any ideas
for a potential solution?

Priyanshu6517, 2023-03-04:
What CPU do you have and what recovery image did you use

v-mod, 2023-03-04:
CPU: Intel i3 10th Gen
Image: volteer

tejasraman, 2023-03-04:
Ok volteer should be the perfect image for this hardware…
Try brunch 110-unstable with a volteer 110 recovery please.

tejasraman, 2023-03-04:
@Priyanshu6517 As I said to you in the other issue, an A6-4400M will never in a million years work with any recovery image.

v-mod, 2023-03-05:
> @v-mod Ok volteer should be the perfect image for this hardware… Try to update to a 110-unstable with the PWA and select 6.1 as the kernel in the settings after the update

ok ill try once I get back from a trip prob 2 or 3 days