sebanc/brunch #1787: Can't enter ChromeOS (settings) on boot options

viltis-projekt, 2023-02-12:
I'm trying to dual boot brunch with win 11 using zork images. I've successfully following the guide but encounter one major problem :
when copying from chromeos.img.grub.txt to grub2win, it gave me  error bellow
[image]

here's what inside chromeos.img.grub.txt
```
menuentry "ChromeOS" --class "brunch" {
	img_path=/chromeos.img
	img_uuid=861a3a08-cee5-48c1-9036-53ea993df28d
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (settings)" --class "brunch-settings" {
	img_path=/chromeos.img
	img_uuid=861a3a08-cee5-48c1-9036-53ea993df28d
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

I could boot into chrome os but stuck at first setup because the wifi (Realtek RTL8852AE WiFi 6 802.11ax PCIe Adapter) doesn't work, and from what I'm understand the fix is inside "ChromeOS (Settings)" on boot option 

please help, and sorry for my bad english

thomas-catt, 2023-02-15:
same issue

wowo-apps, 2023-02-19:
Same issue. ChromeOS dualboot Windows 10.

```
Input file open error

(loop,12)/efi/boot/settings.cfg

Function = BaseArrayRead
```

[Brunch r109 stable 20230201](https://github.com/sebanc/brunch/releases/tag/r109-stable-20230201)
[Rammus 109](
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_15236.80.0_rammus_recovery_stable-channel_mp-v2.bin.zip)
Grub2Win Version 2.3.5.8 Build 1548
Windows 10 Pro 22H2 19045.2604

thomas-catt, 2023-02-20:
> 

> Same issue. ChromeOS dualboot Windows 10.
>
> ```
> Input file open error
>
> (loop,12)/efi/boot/settings.cfg
>
> Function = BaseArrayRead
> ```
>
> [Brunch r109 stable [...]

I ignored the error and everything was working fine.

viltis-projekt, 2023-02-21:
> I ignored the error and everything was working fine.

can you enter brunch config from grub  boot options?

thomas-catt, 2023-02-22:
> can you enter brunch config from grub boot options?

yes, nothing wrong with that.

viltis-projekt, 2023-03-06:
fixed in latest Brunch r110 stable 20230305

closed 2023-03-06 by viltis-projekt