sebanc/brunch #1794: ChromeOS recovery for  Intel N5105?

MonsterXS, 2023-02-27:
ChromeOS recovery for  Intel N5105?

bybenjamin, 2023-03-01:
same question

tejasraman, 2023-03-04:
I'd recommend dedede (as N5105 is Jasper Lake) @MonsterXS @bybenjamin

MonsterXS, 2023-03-04:
> 我推荐dedede（因为N5105是Jasper Lake）@MonsterXS @bybenjamin

Thank you for your answer. I chose hatch because I tested it. It has better compatibility

bybenjamin, 2023-03-04:
> > 我推荐dedede（因为N5105是Jasper Lake）@MonsterXS @bybenjamin
> 
> Thank you for your answer. I chose hatch because I tested it. It has better compatibility

Android apps crash or hang?
How many gb ram is left idle while android app is running

tejasraman, 2023-03-04:
@bybenjamin shat image did you choose?

bybenjamin, 2023-03-04:
> @bybenjamin what image did you choose?

I tried a lot of images ( hatch,dedede,rammus,octopus).  I've had crashes on all android apps except Fydeos.

tejasraman, 2023-03-04:
Does this seem similar to the problem described in #1799 to you?

bybenjamin, 2023-03-04:
> Does this seem similar to the problem described in #1799 to you? I have no Jasperlake hardware so I can't test this myself.
> 
> Also can you please open diagnostics and an android app at the same time? I need to know if the CPU spikes when you use an Android app.
> 
> Instead of using arc++ (like previously), we now use arcvm for android 11 on all modern chrome os recoveries. The way FydeOS implements it, the Android subsystem in Fyde still uses arc++ and android 9.

Yes. this is my post

tejasraman, 2023-03-04:
I don't know why this is.. @sebanc do you know?

lopatkaboli, 2023-12-11:
@bybenjamin 
@tejasraman 
Hi, i have chinese mini pc on n5105 and tried version which suggested by MonsterXS. For me it works fine. 
Used last recovery and last branch.

tejasraman, 2023-12-12:
> @bybenjamin 
> @tejasraman 
> Hi, i have chinese mini pc on n5105 and tried version which suggested by MonsterXS. For me it works fine. 
> Used last recovery and last branch.

Sure that dedede doesn't work? It's technically designed for Jasperlake