sebanc/brunch #1797: Brunch works on StoneyWidge and 2nd gen and up

tejasraman, 2023-03-02:
@sebanc I tested Brunch r109-stable and [this unstable build ](https://github.com/sebanc/brunch-unstable/releases/tag/r110-unstable-20230224) on SandyBridge and StoneyRidge and they work fine. Is there a reason why the readme no longer lists these as supported platforms?

sebanc, 2023-03-02:
Which chromeos recovery image / version did you use for the SandyBridge test ? Is the android subsystem fucntional ?

tejasraman, 2023-03-02:
I used rammus. Android and crostini are functional. I'll send screenshots later if you'd like.

sebanc, 2023-03-02:
Devices with Intel Gen below 4 stopped working a while ago (around r101) due to some incompatible cpu instructions used by ChromeOS, it is possible that something changed again recently. What are your device's specs ?

(Regarding AMD StoneyRidge, the number of devices that work with grunt recovery image is so limited that I removed it completely from compatibility)

tejasraman, 2023-03-02:
Core i5 2435M

tejasraman, 2023-03-02:
I'll send screenshots soon (in less than an hour.

tejasraman, 2023-03-03:
It's booting but **what's the MacBook T2 kernel?** (Using 5.15 for my test as that's what I used last time)

tejasraman, 2023-03-03:
Nevermind, I have a login loop with nami. Trying rammus or coral next?

tejasraman, 2023-03-03:
Ok trying rammus with 5.4 right now...

tejasraman, 2023-03-03:
Looking through for rammus and nami there are illegal opcode errors (missing instructions)

Coral works surprisingly

tejasraman, 2023-03-03:
Ok cut that, Coral doesn't work. It seemed like it did... It works on IvyBridge but not Sandy Bridge (tbh rammus works on ivy too)

closed 2023-03-19 by tejasraman