sebanc/brunch #1803: Brunch R110 + Rammus 110.0.5481.181 - Can't Switch Kernels

onyxsolo, 2023-03-08:
It seems I can no longer switch kernels after updating to brunch R110 and Rammus' build 110.0.5481.181.   If I switch kernels brunch appears to build them like normally during the first subsequent boot. However, the kernel version always shows 5.10.153 under Android in chrome.  I've even tried a fresh install and it only seems to happen under Android 11. Android 9 builds are fine.  

I tried looking through the partitions and by comparison It looks like the brunch kernels aren't actually be moved so ChromeOS might be using the Recovery's stock ChromeOS Kernel not Brunch's.  I'm doing everything I normally had been doing for previous Brunch/ChromeOS updates. 

The only thing I can tell changed is Android 11.  It runs but I get micro stutters and overall performance isn't as good compared to the Sebanc Brunch kernels.  Is there something different I should be doing to switch kernels on Android 11 recoveries?

I'm sorry if my terminology references were bad because I'm not very familiar with linux/android/chrome OS

tejasraman, 2023-03-19:
@onyxsolo where are you seeing the kernel info? With `uname -r` in a terminal or somewhere else?

onyxsolo, 2023-03-22:
I looked under.    system->about device->Android Version->Kernel Version

tejasraman, 2023-03-22:
That's a different kernel version..... That's for the arcvm (android vm). Try `uname -r` in a shell (to get to a shell, press Ctrl+alt+t and type `shell`)

closed 2025-12-29 by github-actions[bot]