sebanc/brunch #1808: Lenovo x13 gen 3

skiv71, 2023-03-17:
Hi,

Just upgraded to this laptop, which I believe has a 13th gen intel CPU.

Anyways, I tried a clean install via volteer image and 110 brunch and various hardware does not function: -
  sound
  wifi
  keyboard backlight (used to be auto).

I had an image circa v103 and installed that and everything works... but I cannot update.

If I update chromeos, i then get a bootloop.

If I rollback and update brunch 1st, on subsequent boot, I cannot even login, due a partition state mismatch, which appears to relate to "credentials".

I'd like to try another chrome recovery image, is there a 13th gen intel version?

skiv71, 2023-03-18:
Ok, I tried a fresh install with a 12th gen ready recovery image and latest brunch, enabled intel_snd...=1 and that's ok. Only thing missing is wifi, related to the ax211 wifi card I believe.

I see people talking about using a backport via grub option but I checked the list and it's not available now?

Gonna be tough using this thing with no wlan card

skiv71, 2023-03-18:
Okay, I have to use the steamdeck kernel to get wifi working... not sure what the differences are there.

My mic isn't working (the laptop have a dual mic array as well as 3.5mm) and this was the case on a t14s gen1 until a brunch update fixed it somehow.

Would be nice to get the mic sorted and be on a proper kernel

tejasraman, 2023-03-18:
Enable the iwlwifi backport and use 6.1 on a 12th gen recovery image like brya. For the mic, could you please try the internal mic fix?

skiv71, 2023-03-18:
How do you enable the backport? It's not in the config wizard

tejasraman, 2023-03-18:
Sorry, it's only in older kernels. AX211 is in-kernel in 5.14+ so try 6.1.

tejasraman, 2023-03-18:
The steam deck one is specifically for testing audio patches for the AMD ACP5x coprocessor.

Please use 6.1 if possible. What problems occur in 6.1?

skiv71, 2023-03-18:
The AX211 wifi card is not detected or enabled.

tejasraman, 2023-03-18:
Can you run `lsmod | grep iwlwifi` in a terminal?

Also, can you run `cat /var/log/dmesg > ~/Downloads/dmesg.txt` and send me the resulting file in Downloads? This is the kernel log and it will show what happened when the driver tried to initialize your card

skiv71, 2023-03-18:
[lsmod.txt](https://github.com/sebanc/brunch/files/11009750/lsmod.txt)
[iwlwifi.txt](https://github.com/sebanc/brunch/files/11009751/iwlwifi.txt)

lsmod doesn't change, but with 6.1 the module fails due to missing file (I think)

tejasraman, 2023-03-18:
Ok could I have the full dmesg?
Just for reference to see if there are any related errors

skiv71, 2023-03-18:
Sure, here it is
 
[dmesg_full.txt](https://github.com/sebanc/brunch/files/11009771/dmesg_full.txt)

tejasraman, 2023-03-18:
Ok, there’s missing firmware…
Can you run “ls -la /lib/firmware | grep iwlwifi-so-a0-gf-a0.pnvm“ to check if the pnvm is there? That seems to be what is causing the Wi-Fi issues.

skiv71, 2023-03-18:
Yes, it's there

skiv71, 2023-03-19:
Actually, my bad... I was in "steamdeck" kernel... and that f/w was present there, but NOT in 6.1

So, I copied it from there and inserted it into 6.1 and booted, wifi working.

What's the upshot? Is this a bug, in that the f/w should be there, but it's absent?

manelscout4life, 2023-03-19:
I have the same wifi issue with my Lenovo Duet i5. I was only able to get wifi working with the steam deck test kernel. Works for me.

tejasraman, 2023-03-19:
@skiv71 that makes a lot of sense. A patch/PR needs to be submitted. I'll look into getting new FW for the kernels and get back to you.