sebanc/brunch #1809: Never used brunch because heard update sucks. But anyway to deal /w it? reinstall?

ccchan234, 2023-03-19:
hi,

finally i think the best thing for me will be a 10inch x86 dual boot win10+chromeos FLEX.

flex, i guess is specifically ENGINEERED NOT to dual boot for specific reasons.

so i turn to brunch for dual boot.

yet i heard even using original google chromebook images, there could be update problem.

may i ask if i use brunch /w flex, it will be the same?

and how to deal /w the update problem?

is it that files are always safe there, and just for each update, i have to manually install once again and its done till next update?

thank you.

sebanc, 2023-03-19:
If your hardware is compatible with chromeos flex and you just want to dualboot, you should use this:
https://github.com/sebanc/linuxloops

As chromeos is not supposed to be dualbooted, there is always a chance it will break but the risks should be much more limited with this other project as it barely changes the OS files.

ccchan234, 2023-03-19:
i found some directions in linuxloop s readme.md

i will try later, and will try to keep win10 on bitlocker.

let's see

thx

closed 2023-03-19 by ccchan234