sebanc/brunch #1810: Can't set secure boot

hikatch72, 2023-03-19:
Hello.
I want to set secure boot on brunch,
How should I set it up?
The grub screen does not come up, and on my computer, I only get red error messages. Is it set in BIOS?

tejasraman, 2023-03-19:
Hi @hikatch72 can you take a pic of the error? You need to install Brunch's secureboot key to get Brunch working with SecureBoot

hikatch72, 2023-03-20:
Hello.  I am using the Celeron N4100 Octopus image, and the model is the Mouse E10 from a Japanese manufacturer.
 How do I install a Secure boot key?  Even if it is enabled in BIOS, it will be like the image attached below.
[image]

tejasraman, 2023-03-20:
@hikatch72 what happens when you click ok? Can you enter the bios?

hikatch72, 2023-03-21:
Press the [OK] button on the attached image to move to the EFI Shell screen. Pressing the ESC button will allow you to enter the BIOS, in which you will find Secure Boot settings.

hikatch72, 2023-03-21:
Pressing the OK button in the attached image will take you to the EFI shell screen. I can enter the BIOS by pressing the ESC button, but how do I install the secure boot key?

tejasraman, 2023-03-21:
@hikatch72 can you run `cd EFI` in the EFI shell, run `cd BOOT`, and run `MokManager.efi` after that.

hikatch72, 2023-03-23:
@tejasraman Thank you for letting me know. I have successfully enabled secure boot. thank you.
I'll make a note of the steps below for anyone who wants to enable secure boot in the brunch framework in the future.
Chrome OS does not support mokutil commands. So you have to start MokManager.efi from EFI Shell.
First, open the BIOS according to the manufacturer's instructions. Then open secure boot option and change to setup mode.
Then change the boot order to set the EFI Shell to boot first, save the changes and reboot. After a while the EFI Shell should start. If it doesn't boot, check the BIOS again.
In the EFI shell, first type fs0: and press enter. Then type cd efi in the command line and press enter. Then type cd boot.
Then type MokManager.efi in the command line and press enter. If Access denied is displayed at this time, check the secure boot setting in BIOS.
If successful, you will see a blue screen. And select OK -> Enroll key from disk -> EFI-SYSTEM -> brunch.der -> select Yes -> reboot Hopefully that's the end of enabling secure boot on brunch. good job for today!
Finally, thanks for answering my question about enabling secure boot properly!
I will close this comment with this. thank you!

closed 2023-03-23 by hikatch72

hikatch72, 2023-03-23:
It seems that my computer didn't have the means to start MokManager for some reason.  Thank you for your advice!

tejasraman, 2023-03-23:
@hikatch72 the advice is only if there is no efi shell option, but thanks :)