sebanc/brunch #1814: You must load the kernel first

valentinciph, 2023-03-20:
I used the dualboot method for Windows.
Used WSL2 is Ubuntu 20.04.5 LTS

menuentry "Brunch" --class "brunch" {
	img_path=/brunch/chromeos.img
	img_uuid=aa008e96-fe31-4a97-8998-5f56089c30b4
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "Brunch settings" --class "brunch-settings" {
	img_path=/brunch/chromeos.img
	img_uuid=aa008e96-fe31-4a97-8998-5f56089c30b4
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

is the code that the script generated. Whenever i try to boot into Brunch via grub the following error shows up:
error: you must load the kernel first
error: unknown filesystem (weird considering i used ntfs for the destination drive)

i have tried multiple installs but no avail.

D0geGames, 2023-05-20:
Same issue here

doandat943, 2023-08-08:
same with me

closed 2025-12-29 by github-actions[bot]